NVIDIA/warp #182: Unstable simulation of joint limits for a hinge joint

slecleach, 2024-02-06:
I've been using warp for a few weeks and it's definitely a very impressive, well executed project!

Currently, I am having trouble simulating a hinge joint using `warp.sim`.
I set the joint range to "-200 45" (in degrees), the rotating link is correctly stopped by the 45deg limit. 
But when I rotate the link towards the -200deg limit the simulation breaks as soon as it gets close to -180deg. 

[image]

Also I have created a minimal working example here:
https://github.com/slecleach/warp/pull/1
You can easily replicate the plot above by running this file:
https://github.com/slecleach/warp/blob/minimal_example/examples/env/env_rotor.py

I am not sure what is causing this, I would appreciate if you could share some insights about this phenomenon.
Thank you!

eric-heiden, 2024-04-11:
Hi Simon,

Thanks for reporting the issue! There seems to be a problem in handling joint coordinates outside [-pi, pi] in general for integrators relying on maximal coordinates (`XPBDIntegrator` (used by default), `SemiImplicitIntegrator`). The `FeatherstoneIntegrator` handles these joint limits correctly, could you work with this for now by setting
```py
integrator_type = IntegratorType.FEATHERSTONE
```
in the `Environment` static variables? We will investigate this issue further in the meantime.

slecleach, 2024-04-21:
Hi Eric, 

Thanks a lot for getting back to me! Yes we'll try the new Featherstone integrator, with minimal coordinates, this type of constraint is somewhat easier to handle. 
I'll also take a look at the XPBD code to see if I can spot the issue. I'll let you know if I find anything ;)

closed 2024-10-11 by shi-eric