NVIDIA/warp #184: Interoperability with PyTorch

TheDevilWillBeBee, 2024-02-16:
Hi,

Thanks for creating this amazing framework. Can you provide a simple example of backpropagation where most of the computation (including the loss function) happens in PyTorch and Warp is only used to implement one of the modules.

Thank you in advance!

sweeneychris, 2024-02-17:
Check out the example here: https://github.com/NVIDIA/warp/blob/main/examples/example_sim_fk_grad_torch.py#L29

This example "injects" the warp functionality as an pytorch autograd function. By using this with .apply() as done [here](https://github.com/NVIDIA/warp/blob/main/examples/example_sim_fk_grad_torch.py#L141) you can freely use this differentiable function inside any larger pytorch system that uses autograd and backprop.

c0d1f1ed, 2024-03-05:
Another Torch interop example was added in https://github.com/NVIDIA/warp/commit/ceb79fcb21c1479a8eb8e5c51c675c82f68c5f14. Thanks Zach!

closed 2024-03-05 by c0d1f1ed