NVIDIA/warp #186: warp 0.13.0 License clarification

traversaro, 2024-02-19:
Hello warp authors, thanks a lot for working on warp and changing license to allow commercial usage (see https://github.com/NVIDIA/warp/releases/tag/v0.13.0). 

Together with @flferretti we are mantaing the conda-forge package of warp, and we wanted to understand how to properly document this new license, in particular using [`SPDX`](https://en.wikipedia.org/wiki/Software_Package_Data_Exchange)

I see that the new license is reference as "NVIDIA Software License" (see for example https://github.com/NVIDIA/warp/blob/9b2a57f7268e8523c24eec1625c6d08548e5c60c/pyproject.toml#L12). However, I am not finding any reference to this "NVIDIA Software License". Is that a "generic"  license used by NVIDIA, or it is something warp-specific? If it is warp specific, could it make sense to call it with a warp-specific name? 

Thanks in advance for the answer!

shi-eric, 2024-03-04:
Hi @traversaro, looking at our internal guidance, I think `SPDX-License-Identifier: LicenseRef-NvidiaProprietary` is the appropriate identifier to use here.

traversaro, 2024-03-04:
Thanks a lot @shi-eric ! I just have a comment on this.

> Hi @traversaro, looking at our internal guidance, I think SPDX-License-Identifier: LicenseRef-NvidiaProprietary is the appropriate identifier to use here.

Based on https://spdx.github.io/spdx-spec/v2.3/other-licensing-information-detected/, ""LicenseRef-"[idstring] where [idstring] is a unique string containing letters, numbers, . and/or -." it seems that the license used by warp is quite warp-specific, and Nvidia employies for its source available software many prorietary licenses, that are all different. Using `LicenseRef-NvidiaProprietary` seems a bit risky to me, as it seems to be highly probably that people will use it to refer to other (but different) licenses used for proprietary Nvidia software. What do you think on this? Could it make sense to use a more warp-specific `idstring`?

shi-eric, 2024-03-04:
Hi @traversaro, I just double-checked with Legal after your last question. I'm told that `LicenseRef-NvidiaProprietary` is used as a "catch-all" and is indeed the correct one to use for Warp.

traversaro, 2024-03-04:
Thanks @shi-eric! Personally I am not sure the use of a "catch-all" code for referring to different license is compatible with Clause 10 of SPDX v2.30 spec, but I guess the Nvidia position is now clear, thanks for the clarification!

closed 2024-03-04 by traversaro