NVIDIA/warp #189: Linear solvers unable to converge

nithinsomu, 2024-02-22:
I am trying to solve a simple 5 X 5 system using gmres in warp. The solver returns nan with and without conditioner. Please find below the code used for the same.

import warp as wp
wp.init()
import numpy as np
from warp.optim.linear import cg, bicgstab, gmres, preconditioner

np.random.seed(123)
n=5
A = np.eye(n)
b = np.random.randn(n)

x0 = wp.array(np.ones(n), dtype=float)
A0 = wp.array(A, dtype=float)
b0 = wp.array(b, dtype=float)
M0 = preconditioner(A0, 'diag')

end_iter, err, atol = gmres(A=A0, b=b0, x=x0, maxiter=10000)

I am running it on cpu. Please tell me if I am doing something wrong.

Thanks 
Nithin

gdaviet, 2024-02-26:
Hi @nithinsomu, I can reproduce the issue, this is due to the current implementation not handling properly a case where the residual converges to zero mid-iteration.  This will be fixed in the next release; in the meantime, perturbing the left-hand-side slightly should avoid the issue.

nithinsomu, 2024-02-26:
Sounds good. Thank you very much.

c0d1f1ed, 2024-03-05:
Fixed by https://github.com/NVIDIA/warp/commit/34254762953848dd25a99fc4b605b7f99dad8440. Thanks Gilles!

closed 2024-03-05 by c0d1f1ed