NVIDIA/warp #191: warp.sparse.bsr_mv report 0-d tensor

FYTalon, 2024-03-01:
version: 0.13.0
In warp.sparse.py file's row 1222:
dim=A.nrow
but warp.sparse store matrix's nrow as a torch tensor. when launch kernel, warp will test len(dim), which will cause 0-d tensor issue.
change the row to
dim=(A.nrow, )
can fix this problem.

gdaviet, 2024-03-02:
Hi @FYTalon, do you have a standalone reproduction example? 
`A.nrow` is of type `int` and `wp.launch` interprets an integer `dim` as a 1d-tuple, so as far as I can tell the two syntaxes should be equivalent

FYTalon, 2024-03-04:
Here is my code:
`import warp as wp
import warp.sparse as wps
import torch
import warp.torch
import numpy as np

wp.init()

a = torch.ones(3, dtype=torch.int32)

n = a.size(0) + a.sum()

mat = wps.bsr_zeros(n, n, block_type=wp.float64)

a = wp.array(shape=n, dtype=wp.float64).zero_()

x = wps.bsr_mv(mat, a)

print(x)
`

and the report is:
`Warp 0.13.0 initialized:
   CUDA Toolkit: 11.5, Driver: 12.2
   Devices:
     "cpu"    | Intel64 Family 6 Model 151 Stepping 2, GenuineIntel
     "cuda:0" | NVIDIA GeForce RTX 3090 (sm_86)
   Kernel cache: C:\Users\Yutao Feng\AppData\Local\NVIDIA\warp\Cache\0.13.0
Traceback (most recent call last):
  File "E:\Project\pienerf\test_mesh_grid.py", line 15, in <module>
    a = wp.array(shape=n, dtype=wp.float64).zero_()
  File "E:\Python\Anaconda\envs\neuProj\lib\site-packages\warp\types.py", line 1548, in __init__
    shape = tuple(shape)
  File "E:\Python\Anaconda\envs\neuProj\lib\site-packages\torch\_tensor.py", line 1022, in __iter__
    raise TypeError("iteration over a 0-d tensor")
TypeError: iteration over a 0-d tensor

Process finished with exit code 1
`

FYTalon, 2024-03-04:
I think it is the problem of using torch tensor as sparse matrix's shape and warp didn't stop me to do so

gdaviet, 2024-06-06:
The bsr matrix constructor has been updated to force cast dimensions to `int` in https://github.com/NVIDIA/warp/commit/9f0b541142b2089fc09a8ef04c462eb0cfe29880

closed 2024-06-06 by gdaviet