NVIDIA/warp #193: Material optimization

honglin-c, 2024-03-11:
Hello! I'm trying to use Warp to do some material optimization experiments. Is there a way to assume the material is uniform and optimize the `k_mu` and `k_lambda` parameters of a tetmesh? 

I tried computing the gradient of `self.model.tet_materials` but all its entries are very close to zero. I am also not sure how I can make the material uniform (e.g., make sure `k_mu` and `k_lambda` are the same for every tetrahedron). 

Thanks!! :)

eric-heiden, 2024-03-14:
Hi Honglin,

Thank you for the question, this would be a great example for differentiable simulation in Warp. We are preparing a material optimization demo in Warp right now and will have it released very soon!

Thanks,
Eric

honglin-c, 2024-03-14:
Hi Eric,

Thank you soooo much for that! Please let me know when it's online. :)

Best,
Honglin

rrzhang139, 2024-11-10:
Hi! Curious about any updates for this demo? Thanks!

shi-eric, 2025-01-28:
@MiguelZamoraM added an example to do this in #419.

closed 2025-01-28 by shi-eric