NVIDIA/warp #196: Cannot use break in a while loop

xuan-li, 2024-03-19:
To reproduce:

```python
import warp as wp

@wp.func
def test_func():
    stop = wp.bool(False)
    while not stop:
        stop = True
        break

@wp.kernel
def test_kernel():
    test_func()

wp.init()
wp.launch(test_kernel, dim=1)
```

Got the following exception:
```
Exception: CUDA kernel build failed with error code 6
```

removing ```break``` works fine.

c0d1f1ed, 2024-04-02:
Thanks for reporting this! A fix will be available in the next release.

KinjalParikh, 2024-04-16:
Similarly `continue` keyword doesn't work either

cadop, 2024-04-16:
I think both break and continue where raised as issues before as well https://github.com/NVIDIA/warp/commit/b57868fac70cce0e585ee0053f038fc3c9e25356

c0d1f1ed, 2024-05-08:
This is fixed as part of [Warp 1.1.0](https://github.com/NVIDIA/warp/releases/tag/v1.1.0)!

closed 2024-05-08 by c0d1f1ed