NVIDIA/warp #198: Alternative tetmesh format

honglin-c, 2024-03-22:
Would it be possible to provide an example where the user reads a tetmesh from a .mesh file? Thank you! :)

honglin-c, 2024-03-22:
I was using the following code (with libigl python binding) to load the tetmesh, which was unsuccessful so far. 

```
  V, T, F = igl.read_mesh("./data/tetmesh/spot.mesh")

  V = Vt.Vec3fArray.FromNumpy(V)
  T = Vt.IntArray.FromNumpy(T)

  builder.add_soft_mesh(
      pos=wp.vec3(0.0, 2.0, 0.0),
      rot=wp.quat_identity(),
      scale=0.1,
      vel=wp.vec3(0.0, 0.0, 0.0),
      vertices=V, 
      indices=T, 
      density=100.0, 
      k_mu=2000.0, 
      k_lambda=2000.0, 
      k_damp=2.0,
      tri_ke=0.0,
      tri_ka=1e4,
      tri_kd=0.0,
      tri_drag=0.0,
      tri_lift=0.0,
  )
```
[image]

mmacklin, 2024-04-02:
Hi @honglin-c, could you provide the .mesh file for this example so we can reproduce?

Thanks,
Miles

samuelpmish, 2024-04-09:
I'm guessing it's this mesh: https://github.com/honglin-c/INSR-PDE/blob/main/elasticity/data/spot.mesh

download link: [spot.zip](https://github.com/NVIDIA/warp/files/14920701/spot.zip)

I tried writing a minimal reproducer with @honglin-c 's approach

```py
import os
import math

import warp as wp
import warp.examples
import warp.sim
import warp.optim
import warp.sim.render

import igl
import scipy as sp
import numpy as np
from pxr import Usd, UsdGeom, Vt

wp.init()

# sim model
builder = wp.sim.ModelBuilder()

V, T, F = igl.read_mesh("/path/to/meshes/spot.mesh")

V = Vt.Vec3fArray.FromNumpy(V)
T = Vt.IntArray.FromNumpy(T)

builder.add_soft_mesh(
    pos=wp.vec3(0.0, 0.0, 0.0),
    rot=wp.quat_identity(),
    scale=2.0,
    vel=wp.vec3(0.0, 0.0, 0.0),
    vertices=V,
    indices=T,
    density=1.0,
    k_mu=2000.0,
    k_lambda=2000.0,
    k_damp=2.0,
    tri_ke=0.0,
    tri_ka=1e-8,
    tri_kd=0.0,
    tri_drag=0.0,
    tri_lift=0.0,
)

# finalize model
model = builder.finalize(requires_grad=False)

model.soft_contact_ke = 2.0e3
model.soft_contact_kd = 0.1
model.soft_contact_kf = 10.0
model.soft_contact_mu = 0.7

model.ground = True

state = model.state(requires_grad=False)

# rendering
renderer = wp.sim.render.SimRenderer(model, "spot.usd")

renderer.begin_frame(0)
renderer.render(state)
renderer.end_frame()

renderer.save()
```

and it seemed to work:
[image]

But I'm very new to using warp, so I might be doing something wrong!

mmacklin, 2024-04-10:
Nice! Thank you @samuelpmish! @honglin-c does this solve your problem?

honglin-c, 2024-04-10:
Wowww thank you so much @samuelpmish!! :)))

honglin-c, 2024-04-10:
Actually, I realized the problem was with the simulation and the scale of the tetmesh. 
For instance, I will have this problem for the code below. But if I change `scale=5.0` to `scale=10.0` in `builder.add_soft_mesh`, the simulation will run without problem. Still trying to figure out what's going wrong here...

honglin-c, 2024-04-10:
This is with `scale = 5.0`:

[image]

Without the simulation (I commented out `example.step()`) , the rendering looks fine: 
[image]

honglin-c, 2024-04-10:
But if I simply change `scale=5.0` to `scale=10.0` in `builder.add_soft_mesh()`, the simulation just runs perfectly! :0 
I'm having a mesh colliding with the ground and then bouncing back in the code: 
[image]
[image]
[image]

mmacklin, 2024-04-11:
This is probably just a classic example of simulation stability being dependent on the scale, e.g.: by scaling everything down then everything gets lighter while stiffness is staying the same, i.e.: the problem got a lot stiffer, which means you need smaller time-steps to be stable.

If you are reducing scale you generally need to reduce the time-step size (dt) or take more substeps as well.

Cheers,
Miles

honglin-c, 2024-04-15:
Thank you so much for the explanations! Now I finally understand. :)

shi-eric, 2024-06-06:
Resolved by @samuelpmish

closed 2024-06-06 by shi-eric