NVIDIA/warp #199: How to use warp?

amdee, 2024-03-24:
Hi, 

I am trying to use Warp in my project. I noticed in the readme that you mentioned Warp is integrated with IsaacGym. 
I was wondering if it's possible to provide guidance on how to use Warp with IsaacGym or show an example of how to use it with IsaacGym. Could someone point me in the right direction? I would really appreciate it.

Thank you.

mmacklin, 2024-04-02:
Hi @amdee, Warp is available and shipped by default in IsaacSim, but not IsaacGym, however because IsaacGym is just a Python package, you can use Warp in conjunction with it by installing them both into your Python environment.

We have some folks using Warp as a way to write reward and loss functions in RL frameworks like IsaacGym. I don't have an example at hand, but you can easily replace many PyTorch computations with Warp and often get a good speed up (please see https://nvidia.github.io/warp/modules/interoperability.html) for more details on how to interop Warp with Torch.

mmacklin, 2024-04-02:
Closing this issue for now, please feel free to continue the discussion on Discord or the Discussions page.

Thanks!
Miles

closed 2024-04-02 by mmacklin

amdee, 2024-04-08:
@mmacklin Thanks for the clarification.