NVIDIA/warp #201: Periodic HashGrid and cylindrical grids

kmarchais, 2024-03-28:
Hello, I would like to request two interrelated features for running physical simulations with Warp.

- Periodic HashGrid:
  - Currently, it seems impossible to implement periodic boundary conditions with HashGrid in one or more dimensions.
  - I propose adding support for periodic boundary conditions to HashGrid, which would be useful for detecting collisions between particles located close to opposite boundaries.

- HashGrid in cylindrical coordinate system:
  - While a normal grid covering the bounds of a cylindrical domain can be used, implementing periodic boundary conditions in the circular direction poses challenges.
  - Introducing a HashGrid specifically designed for cylindrical coordinate systems would simplify handling periodic boundary conditions in such scenarios.

If any of these features are already available or manageable, I would greatly appreciate any guidance on how to utilize them. Otherwise, please let me know if further clarification or assistance is required.

mmacklin, 2024-04-10:
Hi @kmarchais, agreed this is very useful for a number of applications - the `wp.hashgrid` right now is tiling periodically, i.e.: if a particle lies on a boundary cell (as specified by the grid dimensions) then the radius query will wrap and return particles from the other side.

Particles are assigned to hash cells by essentially `cell_coord = x%grid_dim` and neighbors are returned from cells that overlap the query radius.

You would just need to make sure that your distance function is also periodic, and that your domain is equal to `grid_dim*radius`. 

I haven't thought about cylindrical coordinate systems, I think it should be possible, the relevant code to look at is here: https://github.com/NVIDIA/warp/blob/main/warp/native/hashgrid.h#L59.

If you have suggestions on how to extend this to cylindrical coordinates that would be welcome.

Cheers,
Miles

kmarchais, 2024-04-22:
Hello, thank you very much for your answer.
I didn't have time to try it yet but that's what I wanted indeed.

For the cylindrical grid, we can imagine a grid like this:

[image]
[image]

The grid would be defined by a curvature radius and its dimensions.
The coordinates would be something like:
```python
x = np.linspace(-0.5, 0.5, N_POINTS)
y = np.linspace(-0.5, 0.5, N_POINTS)
z = np.linspace(-0.5, 0.5, N_POINTS)

x, y, z = np.meshgrid(x, y, z)

# make periodic
unit_theta = dim[1] / curvature_radius
n_repeat_to_full_circle = int(2 * np.pi / unit_theta)
unit_theta = 2 * np.pi / n_repeat_to_full_circle

rho = x + curvature_radius
theta = y * unit_theta

X = rho * np.cos(theta)
Y = rho * np.sin(theta)
Z = z
```