NVIDIA/warp #205: `model.joint_armature` is not set correctly

etaoxing, 2024-04-07:
[`model.joint_armature`](https://github.com/NVIDIA/warp/blob/v1.0.2/warp/sim/integrator_featherstone.py#L1853) is used inside `FeatherstoneIntegrator.simulate(...)`.

It is not being set correctly inside `parse_mjcf()`, `parse_urdf()`, or `parse_usd()`, since it is not passed to `builder.add_joint()`, so it defaults to 0.01.

MJCF: https://github.com/NVIDIA/warp/blob/v1.0.2/warp/sim/import_mjcf.py#L323-L332

URDF: https://github.com/NVIDIA/warp/blob/v1.0.2/warp/sim/import_urdf.py#L442-L448

USD: https://github.com/NVIDIA/warp/blob/v1.0.2/warp/sim/import_usd.py#L437-L445

eric-heiden, 2024-04-12:
Thanks @etaoxing for reporting these issues! We will add an `armature` argument to the `ModelBuilder.add_joint_...()` methods and apply this settings in the MJCF, URDF, and USD importers in the next release.

shi-eric, 2024-06-06:
I believe this was added by @eric-heiden in 302b79e37725ad46eed4b49b0ba6d4d5160e0dda. Please reopen the issue of the issue is not fixed. Thanks!

closed 2024-06-06 by shi-eric