NVIDIA/warp #211: [feature request] matrix to transform and back

jc211, 2024-04-30:
It would be nice to have methods that convert between a 4x4 matrix to a transform and back. Currently, there a lot of intermediate steps that need to be done to convert between the two.

eric-heiden, 2024-05-01:
Thank you for the suggestion! We will add these functions.

Irene-123, 2024-08-11:
Hi @eric-heiden 
I went through this issue and looking to contribute here
I was hoping if this is an priority task and I could take it up as my good-first-issue?
Lmk if there's something else I can work upon, thanks!

eric-heiden, 2024-08-12:
Hi @Irene-123,

Thank you for offering to contribute! This is indeed a good first issue you could help with! You can find our contributions guidelines [here](https://github.com/NVIDIA/warp/blob/main/CONTRIBUTING.md) to get started.

Thanks,
Eric

Irene-123, 2024-08-12:
Hi @eric-heiden Yes, I have joined the discord, will provide more updates over there to you

shi-eric, 2025-02-23:
@jc211: @christophercrouzet just added this in 16459440de446f770f0b299c57ef04573ab4dec6. It'll be included in the 1.7 release, or you can try it out sooner in the nightly wheels.

closed 2025-02-23 by shi-eric