NVIDIA/warp #215: Can I use implementation details from hashgrid in another project

SteffenPL, 2024-05-03:
Dear NVIDIA-Warp Team,

I'm working on a Julia package for spatial hash tables ([SpatialHashTables.jl](https://github.com/SteffenPL/SpatialHashTables.jl)). 

After reading your code in [warp/native/hashgrid.h](https://github.com/NVIDIA/warp/blob/3ed2ceab824b65486c5204d2a7381d37b79fc314/warp/native/hashgrid.h), [hashgrid.cpp](https://github.com/NVIDIA/warp/blob/3ed2ceab824b65486c5204d2a7381d37b79fc314/warp/native/hashgrid.cpp) and [hashgrid.cu](https://github.com/NVIDIA/warp/blob/main/warp/native/hashgrid.cu) it is hard to unsee certain tricks. 

Is it OK to use these three files as an inspiration for the Julia package? 
[I am in particular aiming at a decent GPU performance via [CUDA.jl](https://github.com/JuliaGPU/CUDA.jl).]

I'm asking in particular in order to satisfy the file's license which states: 
```
/** Copyright (c) 2022 NVIDIA CORPORATION.  All rights reserved.
 * NVIDIA CORPORATION and its licensors retain all intellectual property
 * and proprietary rights in and to this software, related documentation
 * and any modifications thereto.  Any use, reproduction, disclosure or
 * distribution of this software and related documentation without an express
 * license agreement from NVIDIA CORPORATION is strictly prohibited.
 */
```

Please let me know if an e-mail is better suited for this question. 

Kind regard,
Steffen

shi-eric, 2024-07-02:
Hi Steffen, I've been trying to get an answer to your question by contacting our Legal department but haven't found anyone willing to provide an answer. Sorry. I can send you an e-mail of the person who is technically responsible for questions regarding the license Warp uses if you want to follow up with them.

closed 2024-07-31 by shi-eric

shi-eric, 2025-03-07:
We're now using the Apache 2.0 license so that should make your usage question a lot more straightforward.