NVIDIA/warp #216: Building from source with conda

jc211, 2024-05-04:
In a conda environment, the current build scripts do not find the appropriate CUDA headers required to compile. Its somewhat difficult to debug since the build scripts are custom.

pculbertson-bdai, 2024-06-25:
@alberthi and I are hitting this same issue -- and made some headway on reproducing/debugging it.

Here's some code to set up a vanilla conda env + try to build warp:
```
conda create -n warp-test python=3.10
conda install -c nvidia/label/cuda-12.2.2 cuda # install cuda toolkit
python build_lib.py # errors due to linking issues
```
We found if you run `conda install -c conda-forge cxx-compiler` it's able to find the headers (sometimes we had to run `export CUDA_PATH=$CONDA_PREFIX`).

However, we still are getting build errors -- specifically, during the clang/llvm portion, we get:
```
warp/_build/host-deps/llvm-project/release-x86_64/include/llvm/Object/SymbolicFile.h:48:30: error: expected ')' before 'PRIxPTR'
   48 |   OS << "(" << format("0x%08" PRIxPTR, D.p) << " (" << format("0x%08x", D.d.a)
```
So it still looks like some of the macros aren't getting linked properly. We can still build/use the library by passing `--no_standalone` to `build_lib.py` but aren't sure if this will cause issues down the line. Thanks!!

c0d1f1ed, 2024-07-02:
I think you're on the right track with `conda install -c conda-forge cxx-compiler`. Can you tell me what version of GCC this installs? You may also need a newer glibc version. Try installing a specific version like `conda install -c conda-forge libstdcxx-ng=9.5`.

traversaro, 2024-10-02:
In case anyone is interested in this, we have some patches and build scripts in https://github.com/conda-forge/warp-lang-feedstock/tree/main/recipe to build warp with conda-forge dependencies.