NVIDIA/warp #220: Interoperability with pytorch geometric

nithinsomu, 2024-05-18:
Just out of curiosity, does warp currently or in near future have plans to support interoperability with torch geometric? This would be useful when we have simulations in unstructured grid and want to use graph neural networks.

mmacklin, 2024-05-26:
Hi @nithinsomu, I'm not particularly familiar with Torch Geometric - can you expand on which particular features would be the most useful? Or is it mostly a matter of data format conversions we should support?

nithinsomu, 2024-05-28:
Hey Miles, 
Thank you for the reply. I was thinking along the lines of supporting the graph data format which is used in pytorch geometric, performing the transformations and may be some of the graph convolution operations if possible. 

Thanks and Regards,
Nithin Somasekharan