NVIDIA/warp #222: The gradient is 0 when using pytorch and warp for joint optimization

huahuo359, 2024-05-23:
I want to train a pytorch model with warp, but the grad is always zero, What should I pay attention to in code implementation?
```python
import warp as wp
import torch
import torch.nn as nn

class MLP(nn.Module):
    def __init__(self):
        super(MLP, self).__init__()
        self.fc1 = nn.Linear(3, 10)
        self.fc2 = nn.Linear(10, 10)

    def forward(self, x):
        x = self.fc1(x)
        x = self.fc2(x)
        return x

wp.init()

@wp.kernel()
def loss(x1: wp.array(dtype=float, ndim=3), x2: wp.array(dtype=float, ndim=3), l: wp.array(dtype=float)):
    tid1, tid2 = wp.tid()
    wp.atomic_add(l, 0, (x1[tid1, tid2, 0] - x2[tid1, tid2, 1]) ** 2.0)

# indicate requires_grad so that Warp can accumulate gradients in the grad buffers
xs = torch.randn(100, 10, 3,  requires_grad=True)
x_gt = torch.ones(100, 10, 10)
l = torch.zeros(1, requires_grad=True)
opt = torch.optim.Adam([xs], lr=0.1)
wp_gt = wp.from_torch(x_gt)
wp_l = wp.from_torch(l)

tape = wp.Tape()

mlp = MLP()
for i in range(500):
    wp.synchronize()
    tape.zero()
    tape.backward(loss=wp_l)  
    opt.step()  
    wp_l.zero_()
    with tape:
        wp_xs = mlp(xs)
        wp_xs = wp.from_torch(wp_xs, requires_grad=True)
        wp.launch(loss, dim=len(xs), inputs=[wp_xs, wp_gt], outputs=[wp_l], device=wp_xs.device)
        print("wp_xs grad: ", wp_xs.grad)
    print(f"{i}\tloss: {l.item()}")
    # if i != 0:
    #     print("xs_torch grad: ", xs_torch.grad)
    

```

mmacklin, 2024-05-26:
Hi @huahuo359,

The `wp.Tape()` can't capture Torch operations, instead of wrapping the MLP in a Warp Tape I think it would be better to wrap your custom loss function in a `torch.autograd.func` as shown here:

https://nvidia.github.io/warp/modules/interoperability.html#example-optimization-using-torch-autograd-function

Hope that helps.

Cheers,
Miles

huahuo359, 2024-05-27:
Thank you very much for your answer，I implemente a demo code like this:
My purpose is to try the possibility of combining torch and warp for optimization
I don't know if it's a problem with my implementation or if the current support for such an operation is not enough.
```python
import warp as wp
import torch
import torch.nn as nn

class MLP(nn.Module):
    def __init__(self):
        super(MLP, self).__init__()
        self.fc1 = nn.Linear(3, 10)
        self.fc2 = nn.Linear(10, 3)

    def forward(self, x):
        x = self.fc1(x)
        x = self.fc2(x)
        return x

wp.init()

@wp.kernel()
def square_kernel(input: wp.array(dtype=float, ndim=3), output: wp.array(dtype=float, ndim=3)):
    tid1, tid2, tid3 = wp.tid() 
    output[tid1, tid2, tid3] = input[tid1, tid2, tid3] + input[tid1, tid2, tid3]

@wp.kernel
def loss_l2(x: wp.array(dtype=float, ndim=3), y: wp.array(dtype=float, ndim=3), res: wp.array(dtype=float)):
    tid1, tid2, tid3 = wp.tid()
    wp.atomic_add(res, 0, warp.abs(x[tid1, tid2, tid3] - y[tid1, tid2, tid3]))
    

class Rosenbrock(torch.autograd.Function):
    @staticmethod
    def forward(ctx, input, gt):
        wp.synchronize_device()
        ctx.input = wp.from_torch(input, requires_grad=True)
        ctx.gt = wp.from_torch(gt)
        
        ctx.loss = wp.zeros(1, requires_grad=True)
        
        wp.launch(
            kernel=loss_l2,
            dim=len(gt),
            inputs=[ctx.input, ctx.gt],
            outputs=[ctx.loss]   
        )
        wp.synchronize_device()
        return wp.to_torch(ctx.loss)
    
    @staticmethod   
    def backward(ctx, adj_loss):
        wp.synchronize_device()
        ctx.loss.grad = wp.from_torch(adj_loss)
        wp.launch(
            kernel=loss_l2,
            dim=len(ctx.gt),
            inputs=[ctx.input, ctx.gt],
            outputs=[ctx.loss],
            adj_inputs=[ctx.input.grad, ctx.gt.grad],
            adj_outputs=[ctx.loss.grad],
            adjoint=True
        )
        
        wp.synchronize_device()
        return (wp.to_torch(ctx.input.grad), None)
    

def apply_square(input_tensor):
    input_tensor = input_tensor.float()
    input_array = wp.from_torch(input_tensor)
    output_array = wp.empty_like(input_array)
    wp.launch(square_kernel, dim=input_tensor.shape, inputs=[input_array, output_array])
    wp.synchronize_device()
    output_tensor = wp.to_torch(output_array)
    return output_tensor

xs = torch.randn(100, 10, 3, requires_grad=True)
x_gt = torch.ones(100, 10, 3)
l = torch.zeros(1, requires_grad=True)
opt = torch.optim.Adam([xs], lr=0.1)

mlp = MLP()

for i in range(5000):
    opt.zero_grad()
    xs_2 = apply_square(xs)

    wp_xs = mlp(xs_2)

    wp_xs.retain_grad()
    loss = Rosenbrock.apply(wp_xs, x_gt)
    loss.backward(torch.ones_like(loss))
    print(loss)
    print(xs.grad)
    print(wp_xs.grad)
    print(xs.grad)
    opt.step()

    if i % 50 == 0:
        print(f"Iteration {i}, Loss: {loss.item()}")

```

closed 2024-05-28 by huahuo359