NVIDIA/warp #229: Ways to print logging information for warp kernel function

RuihanGao, 2024-06-03:
Thanks for the great work! 
I may miss it in the documentation, but is it possible to print/log any debugging/error messages in warp kernel functions with `@wp.kernel` decorator? 
Hint/instructions would be greatly helpful, thanks!

shi-eric, 2024-06-03:
Hey @RuihanGao, are you looking for [wp.printf()](https://nvidia.github.io/warp/modules/functions.html#warp.printf)?

Here's a basic example: https://nvidia.github.io/warp/debugging.html#printing-values

RuihanGao, 2024-06-04:
> Hey @RuihanGao, are you looking for [wp.printf()](https://nvidia.github.io/warp/modules/functions.html#warp.printf)?
> 
> Here's a basic example: https://nvidia.github.io/warp/debugging.html#printing-values

That's exactly what I was looking for, thanks a lot for pointing it out!

closed 2024-06-04 by RuihanGao