NVIDIA/warp #230: mjcf imports

robots4us, 2024-06-04:
Hi! I am trying to import some robots from the mujoco menagrie. Eg. the shadow hand -> https://github.com/google-deepmind/mujoco_menagerie/blob/main/shadow_hand/right_hand.xml

However, I keep getting errors:

count_contact_points: unsupported geometry type combination 3 and 7
Unsupported geometry pair in collision handling

Can you recommend a quick fix? Thank you for this amazing simulator by the way. :)

eric-heiden, 2024-06-06:
Hi @robots4us,

These geometry type numbers refer to the `GEO_...` constants, so 3 and 7 refer to `GEO_CYLINDER` and `GEO_PLANE`, see [model.py](https://github.com/NVIDIA/warp/blob/main/warp/sim/model.py#L38-L46).

Since we don't have rigid-body contacts implemented for cylinders yet, I would recommend swapping them out for capsules in the MJCF file for now.

robots4us, 2024-06-07:
That worked. Thanks! :+1:

robots4us, 2024-06-07:
On closer inspection, the XML is imported fine, but some joint axes and joint limits are incorrect. In particular, anything that is declared at the beginning of the XML and later does not appear to have been imported correctly. I wrote a script to read the .XML and change the warp.sim.Model. Is there something I might be doing incorrectly or any other workarounds?