NVIDIA/warp #234: [DOCS] Basic Example Missing wp.init() call

tchild, 2024-06-13:
## Category

- [x ] Report an error in the documentation.
- [ ] Request for something to be documented.
- [ ] Request for a new [example](https://github.com/NVIDIA/warp/tree/main/warp/examples) to illustrate how to do something.
- [ ] Other (please explain)

## Description

christophercrouzet, 2024-06-13:
Hi @tchild, `wp.init()` shouldn't be required anymore in the latest release of Warp, which is why it was removed from the documentation. Are you running an older version than 1.2.0 by any chance?

tchild, 2024-06-13:
With wp_init() removed:

C:\Users\timch\PycharmProjects\Nvidia_Warp_test\venv\Scripts\python.exe C:\Users\timch\PycharmProjects\Nvidia_Warp_test\examples\random_3d_vectors.py 
Traceback (most recent call last):
  File "C:\Users\timch\PycharmProjects\Nvidia_Warp_test\examples\random_3d_vectors.py", line 20, in <module>
    points = wp.array(np.random.rand(num_points, 3), dtype=wp.vec3)
  File "C:\Users\timch\PycharmProjects\Nvidia_Warp_test\venv\lib\site-packages\warp\types.py", line 1626, in __init__
    self._init_from_data(data, dtype, shape, device, copy, pinned)
  File "C:\Users\timch\PycharmProjects\Nvidia_Warp_test\venv\lib\site-packages\warp\types.py", line 1781, in _init_from_data
    device = warp.context.runtime.get_device(device)
AttributeError: 'NoneType' object has no attribute 'get_device'
Warp 1.2.0 initialized:
   CUDA Toolkit 11.8, Driver 12.3
   Devices:
     "cpu"      : "Intel64 Family 6 Model 141 Stepping 1, GenuineIntel"
     "cuda:0"   : "NVIDIA GeForce RTX 3070 Laptop GPU" (8 GiB, sm_86, mempool enabled)
   Kernel cache:
     C:\Users\timch\AppData\Local\NVIDIA\warp\Cache\1.2.0

Process finished with exit code 1

tchild, 2024-06-13:
With wp.init() called

C:\Users\timch\PycharmProjects\Nvidia_Warp_test\venv\Scripts\python.exe C:\Users\timch\PycharmProjects\Nvidia_Warp_test\examples\random_3d_vectors.py 
Warp 1.2.0 initialized:
   CUDA Toolkit 11.8, Driver 12.3
   Devices:
     "cpu"      : "Intel64 Family 6 Model 141 Stepping 1, GenuineIntel"
     "cuda:0"   : "NVIDIA GeForce RTX 3070 Laptop GPU" (8 GiB, sm_86, mempool enabled)
   Kernel cache:
     C:\Users\timch\AppData\Local\NVIDIA\warp\Cache\1.2.0
Module __main__ 8267b0c load on device 'cuda:0' took 20.24 ms
[0.74909157 1.1493175  1.3429637  ... 1.0309446  0.9536935  1.1456277 ]

Process finished with exit code 0

christophercrouzet, 2024-06-13:
Right, this one is in fact a bug! The fix for it has actually been committed a couple of days ago (see commit [c3e2195](https://github.com/NVIDIA/warp/commit/c3e2195466a2ce00d3ccc548686c3109e905a183) and has just been merged now, so it is available in the branch `main` and will also be available in the next release.

Sorry for the inconvenience!

tchild, 2024-06-13:
@christophercrouzet  thanks for the  prompt update.  It's not a problem for me now, but it throws off the any newbie trying to run the first example of code in the doc. I will hold off from submitting  at PR to fix the doc as its not required.

closed 2024-06-13 by tchild

c0d1f1ed, 2024-06-14:
The fix was released as part of the [v1.2.1](https://github.com/NVIDIA/warp/releases) package (also available through PyPI). Thanks Chris!