NVIDIA/warp #237: Can't compute jacobian with wp.struct?

supertan0204, 2024-06-16:
Hi,

I'm using Warp's autodiff to compute a **jacobian**, following the tutorial from https://nvidia.github.io/warp/modules/differentiability.html#. I realize that if the data are enclosed in a `wp.struct` then the jacobian cannot be computed. See a simple example code block below:
```
import numpy as np
import warp as wp
import torch
wp.init()

@wp.struct
class my_class:
    E: wp.array(dtype=wp.float32)
    
    lam: wp.array(dtype=wp.float32)
    
state = my_class()
state.lam = wp.zeros(3, dtype=wp.float32, device="cuda:0")

E_np = np.array([1,2,3], dtype=np.float32)
state.E = wp.from_numpy(E_np, device='cuda:0', dtype=wp.float32, requires_grad=True)

@wp.kernel
def test(state: my_class):
    p = wp.tid()
    state.lam[p] = state.E[p] / 3.0 # a very simple computation

# compute jacobian following https://nvidia.github.io/warp/modules/differentiability.html#
tape = wp.Tape()
input_dim = output_dim = 3
with tape:
    wp.launch(test, dim=3, inputs=[state])

for output_index in range(output_dim):
    select_index = torch.zeros(output_dim)
    select_index[output_index] = 1.0
    e = wp.array(select_index, dtype=wp.float32)
    tape.backward(grads={state.lam: e})
    q_grad_i = tape.gradients[state.E] 

    # print the gradient
    print(q_grad_i.numpy()) # this gives all zero....

    tape.zero()
```
This gives **all-zero** gradients:
```
Warp 0.10.1 initialized:
   CUDA Toolkit: 11.5, Driver: 12.2
   Devices:
     "cpu"    | x86_64
     "cuda:0" | NVIDIA GeForce RTX 3090 (sm_86)
   Kernel cache: [/home/xiyang/.cache/warp/0.10.1](https://file+.vscode-resource.vscode-cdn.net/home/xiyang/.cache/warp/0.10.1)
Module __main__ load on device 'cuda:0' took 0.58 ms
[0. 0. 0.]
[0. 0. 0.]
[0. 0. 0.]
```

However, if I unpack the `wp.struct` and load the arrays separately, the gradient was correct (not accidentally):
```
import numpy as np
import warp as wp
import torch
wp.init()

lam = wp.zeros(3, dtype=wp.float32, device="cuda:0")

E_np = np.array([1,2,3], dtype=np.float32)
E = wp.from_numpy(E_np, device='cuda:0', dtype=wp.float32, requires_grad=True)

@wp.kernel
def test(lam: wp.array(dtype=wp.float32), E: wp.array(dtype=wp.float32)):
    p = wp.tid()
    lam[p] = E[p] / 3.0 # a very simple computation

# compute jacobian following https://nvidia.github.io/warp/modules/differentiability.html#
tape = wp.Tape()
input_dim = output_dim = 3
with tape:
    wp.launch(test, dim=3, inputs=[lam, E])
for output_index in range(output_dim):
    select_index = torch.zeros(output_dim)
    select_index[output_index] = 1.0
    e = wp.array(select_index, dtype=wp.float32)
    tape.backward(grads={lam: e})
    q_grad_i = tape.gradients[E] 

    # print the gradient
    print(q_grad_i.numpy()) # this gives the correct result
    
    tape.zero()
``` 
And this gives the correct results:
```
Warp 0.10.1 initialized:
   CUDA Toolkit: 11.5, Driver: 12.2
   Devices:
     "cpu"    | x86_64
     "cuda:0" | NVIDIA GeForce RTX 3090 (sm_86)
   Kernel cache: [/home/xiyang/.cache/warp/0.10.1](https://file+.vscode-resource.vscode-cdn.net/home/xiyang/.cache/warp/0.10.1)
Module __main__ load on device 'cuda:0' took 200.11 ms
[0.33333334 0.         0.        ]
[0.         0.33333334 0.        ]
[0.         0.         0.33333334]
```

supertan0204, 2024-06-16:
I tried to add `requires_grad = True` to the dummy array (`state.lam`, in the above codes) and the jacobian is no longer zero this time. However, the jacobian, as a form of matrix, is still not correct:
```
import numpy as np
import warp as wp
import torch
wp.init()
@wp.struct
class my_class:
    E: wp.array(dtype=wp.float32)
    
    lam: wp.array(dtype=wp.float32)

state = my_class()
# Here add requires_grad=True to state.lam
state.lam = wp.zeros(3, dtype=wp.float32, device="cuda:0", requires_grad=True)

E_np = np.array([1,2,3], dtype=np.float32)
state.E = wp.from_numpy(E_np, device='cuda:0', dtype=wp.float32, requires_grad=True)

@wp.kernel
def test(state: my_class):
    p = wp.tid()
    state.lam[p] = state.E[p] / 3.0 # a very simple computation

# compute jacobian following https://nvidia.github.io/warp/modules/differentiability.html#
tape = wp.Tape()
input_dim = output_dim = 3
with tape:
    wp.launch(test, dim=3, inputs=[state])
for output_index in range(output_dim):
    select_index = torch.zeros(output_dim)
    select_index[output_index] = 1.0
    e = wp.array(select_index, dtype=wp.float32)
    # pass input gradients to the output buffer to apply selection
    tape.backward(grads={state.lam: e})
    q_grad_i = tape.gradients[state.E] 

    # print the gradient
    print(q_grad_i.numpy()) # this gives all zero....

    
    tape.zero()
```
This gives a output of:
```
Warp 0.10.1 initialized:
   CUDA Toolkit: 11.5, Driver: 12.2
   Devices:
     "cpu"    | x86_64
     "cuda:0" | NVIDIA GeForce RTX 3090 (sm_86)
   Kernel cache: /home/xiyang/.cache/warp/0.10.1
Module __main__ load on device 'cuda:0' took 0.97 ms
[0. 0. 0.]
[0.         0.33333334 0.        ]
[0.         0.33333334 0.        ]
```

gdaviet, 2024-06-17:
Hi @supertan0204 , I see you're using version "0.10.1", can you try with a more recent one? This issue seems to have been fixed since version 1.0.0 at least

supertan0204, 2024-06-17:
Hi @gdaviet, I've updated my warp version to latest "1.2.1" and find it still gives all zero jacobian by running the first code block.
```
Warp 1.2.1 initialized:
   CUDA Toolkit 11.8, Driver 12.2
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce RTX 3090" (24 GiB, sm_86, mempool enabled)
   Kernel cache:
     [/home/xiyang/.cache/warp/1.2.1](https://file+.vscode-resource.vscode-cdn.net/home/xiyang/.cache/warp/1.2.1)
Module __main__ 5eb0f1e load on device 'cuda:0' took 0.42 ms
[0. 0. 0.]
[0. 0. 0.]
[0. 0. 0.]
```

gdaviet, 2024-06-17:
Ah, I've only tested the block from your second comment (with `requires_grad = True`) -- this one should now be working. But I agree that since you're passing custom grad arrays, one could expect the code to work withouyt requires_grad on `state.lam` (or at least get a warning)

supertan0204, 2024-06-17:
I see block two works now, thanks.

I suppose there should be some warning when a user trying to compute gradients using `wp.struct` whithout setting *dummy variables'* (`state.lam` in this example) `requires_grad`** to `True`? As a convention (e.g. in Pytorch), I think `requires_grad=True` is not needed for these variables.

> Ah, I've only tested the block from your second comment (with `requires_grad = True`) -- this one should now be working. But I agree that since you're passing custom grad arrays, one could expect the code to work withouyt requires_grad on `state.lam` (or at least get a warning)

closed 2024-07-03 by supertan0204