NVIDIA/warp #241: [QUESTION] Very high memory usage in warp.mesh.ModelBuilder.finalize()

knauth, 2024-06-19:
I'm running a simulation which involves thousands of bodies interacting. Each body has ~20 collision spheres. I can add these, albeit slowly, using add_body and add_shape_sphere. However, when calling builder.finalize(), memory usage is untenably high; more than 150 GB for ~2k bodies. I think this is because the list of contact pairs is very long.

Is there a way to optimize finalize() to avoid this quadratic growth? In libraries like Chrono the contact pairs are trimmed intelligently, does warp have similar features? I really like the syntax, docs, and native GPU support Warp provides, so it'd be great to be able to perform many-body sims efficiently.

eric-heiden, 2024-06-19:
Hi @knauth,

Just to make sure - are you running your simulation with the most current Warp version? We improved the allocation of contact points last week that improves the memory usage: https://github.com/NVIDIA/warp/commit/95937adcad70d48e38019bf3426afe24984f26e8

We will look into a more efficient broadphase collision detection soon and also aim to reduce the memory usage further for these contact scenarios where all bodies can interact with each other. For now, the number of contacts per environment has been limited via collision groups and filters (which are applied statically for the simulation at the beginning when calling `ModelBuilder.finalize()`).

knauth, 2024-06-19:
Just built the lang from the repo, there was some improvement but it's still quite memory-intensive. 

Is there any improvement to be had from transitioning to collision spheres to full-body triangle meshes, or another type of collision shape? We are representing many extended tori for a chainmail simulation.

Warp is really cool but we're probably going to look into PhysX for this simulation since collision detection is much more mature there and still GPU-native. Looking forward to what we see from Warp in the future, though- we run a computational geometry pipeline that's really going to benefit from the ability to compile Python to the GPU. Thanks for the help Eric!

knauth, 2024-06-19:
Can you estimate the person-hours that would go into implementing such a broadphase collision algorithm? If we can help with an upstream contribution, my company can probably devote some time to this, although we're not experts in rigid-body problems.

knauth, 2024-06-23:
Hey @eric-heiden, we took some time this week to play with things and implemented what was, to us, the most natural functionality change to support many-body sims. By removing the creation of the list of valid cartesian products (which grew quadratically), we removed the massive memory requirements of ModelBuilder.finalize(). To replicate that functionality, we tweaked the broadphase collision detection to operate as a 2D kernel over the list of shapes, with the validity of the collision determined at runtime by the kernel itself. Everything else was untouched.

In our testing, this allowed for ~2k bodies of 20 shapes each in under 5GB of VRAM, with ~11s per 1/60th of a second timestep on a 3060. We observed a ~1-2ms performance hit for smaller sims versus the build-time cartesian method.

Right now this removes support for filters and collision groups, but they could be implemented in the kernel for a performance hit. The code (see [our fork](https://github.com/opensuit-labs/warp/tree/feat/many-body-sims)) is also a mess; our first goal was to prove the validity of the method. Over the next week we'll be polishing this off and separating the functionality.

This is well suited for our use case, i.e. the following scenario:
- Many bodies
- No collision groups
- No filters (all bodies could collide under certain circumstances, but are unlikely to)

Because it's already slow to simulate so many bodies, we're okay with the broadphase performance hit to make this work.

**Would the Warp team be open to a PR adding this functionality?** We're going to be developing it on our fork but would like to upstream this feature (once polished and well-integrated) to support similar use cases. We are open to suggestions on architecture; I like the idea of composition/inheritance to support adding, say, a CartesianModelBuilder and RuntimeModelBuilder, with changes to the collision and integration kernels to minimize code duplication there.

eric-heiden, 2024-06-24:
Hey @knauth, thank you for these contributions!

The results are very interesting how the 2D broadphase kernel can help in those scenarios.

It would be great to get collision groups supported as well, which should be straightforward by turning the `shape_collision_group` into a Warp array as member of the Model and adding the collision group logic to the broadphase kernel (which shouldn't incur much overhead). For collision filter pairs I made a quick test on how the shape index pairs could be encoded in a bool array of dimension $`O(n\log(n))`$:

```py
import warp as wp

shape_count = 5
k = int(wp.log2(float(shape_count + 1)))

@wp.func
def pack_ints(a: int, b: int):
    return (a << k) | b

@wp.kernel
def match_pairs(
    pair_filter: wp.array(dtype=bool),
):
    a, b = wp.tid()
    key = pack_ints(a, b)
    wp.printf("key[%d, %d] = %d\n", a, b, pair_filter[key])

filter_pairs = set([(1, 2), (0, 3), (2, 1), (3, 0)])
capacity = shape_count << k
filter_pairs_array = [False] * capacity
for a, b in filter_pairs:
    filter_pairs_array[pack_ints(a, b)] = True

filter_pairs_array = wp.array(filter_pairs_array, dtype=bool)

wp.launch(match_pairs, dim=(shape_count, shape_count), inputs=[filter_pairs_array])
```

I think for now you can work with your fork and I can do some testing with adding the collision group and pair-wise filtering to your broadphase approach (since you don't need it right now) to better understand the performance difference to the current broadphase handling for different simulation use cases. If the performance is not impacted significantly this could become the default behavior.

knauth, 2024-06-24:
@eric-heiden Sounds good- we're happy with this feature set at the moment. Excited to see how filters and groups play with this strategy. For ground contacts, we've left them as-is with build time generation at the moment, because the list only grows linearly. Depending on performance requirements, the ground could likely be integrated into the list of shapes in the 2D broadphase, which would reduce some of the code complexity.

On the same note- is there a performance issue with calling one kernel from another? I noticed the code in count_contact_points is very similar to the first phase of broadphase_collision_pairs. Is it worth consolidating?

closed 2024-10-11 by shi-eric