NVIDIA/warp #243: Support for vector operations with length greater than 4

ricetwice, 2024-06-21:
Hello,

I encountered an issue while using the Warp framework. It seems that the framework does not support basic operations (like addition) for vectors with a length greater than 4. Below is a minimal reproducible example of the code and the error message I received.

**Code**:
``` python
import warp as wp

vec5f = wp.types.vector(length=5, dtype=wp.float32)

a = vec5f(1.0)
b = vec5f(2.0)

# c = wp.cw_mul(a, b)
c = a + b
print(c)
```
**Error Message**:
```
Warp 1.2.1 initialized:
   CUDA Toolkit 11.8, Driver 12.5
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce GTX 1060 6GB" (6 GiB, sm_61, mempool enabled)
   Kernel cache:
     /home/tworice/.cache/warp/1.2.1
Traceback (most recent call last):
  File "/home/tworice/Projects/warp_test/test_cwmul.py", line 9, in <module>
    c = a+b
  File "/home/tworice/.local/lib/python3.10/site-packages/warp/types.py", line 207, in __add__
    return warp.add(self, y)
  File "/home/tworice/.local/lib/python3.10/site-packages/warp/context.py", line 206, in __call__
    raise RuntimeError(
RuntimeError: Couldn't find a function 'add' compatible with the arguments 'vec_t, vec_t'
```

Environment:

- Warp version: 1.2.1
- CUDA Toolkit: 11.8
- CUDA Driver: 12.5
- GPU: NVIDIA GeForce GTX 1060 6GB
- Operating System: Ubuntu 22.04.4 LTS
- Python version: 3.10

It would be great if support for vectors with length greater than 4 could be added, or if there is a workaround, please let me know.

Thank you!

ricetwice, 2024-06-27:
Hi there,

The issue has been resolved. I found that vector operations with lengths greater than 4 only work correctly within warp kernels.

closed 2024-06-27 by ricetwice

christophercrouzet, 2024-06-30:
I @ricetwice, thanks for reporting this issue and sorry for any inconvenience caused!

You're correct, such operations on non-predefined types, like `vec5`, are currently only supported inside kernels, but not with Warp's Python runtime.

We have not tackled this work to implement this feature yet but it's on our roadmap—did you have a need for it?