NVIDIA/warp #250: [QUESTION] Moving mesh during physic simulation

SangHunHan92, 2024-07-04:
Thanks for great work !

I want to simulate a cloth for a moving object, exactly like this question.
https://github.com/NVIDIA/warp/issues/160

But I couldn't find updated + refit() in the link below:
https://github.com/NVIDIA/warp/blob/main/warp/sim/model.py#L256

But, I found simillar functions during search "refit":
https://github.com/NVIDIA/warp/blob/bfbd0258479ea1378788c2ff5bdbbdc1da385612/warp/types.py#L2997-L3004
https://github.com/NVIDIA/warp/blob/bfbd0258479ea1378788c2ff5bdbbdc1da385612/warp/examples/core/example_mesh.py#L116-L129

Has the function to update mesh been moved?

Also, if I combine this properly with the example below, can I get it to work as I intended?
https://github.com/NVIDIA/warp/blob/main/warp/examples/sim/example_cloth.py

SangHunHan92, 2024-07-05:
Now I can deform the mesh using wp.launch. It's working well !!

```
@wp.kernel
def deform(positions: wp.array(dtype=wp.vec3), new_positions: wp.array(dtype=wp.vec3)):    
    tid = wp.tid()
    positions[tid] = new_positions[tid]

wp.launch(kernel=deform, dim=len(self.mesh.points), inputs=[self.mesh.points, self.animations[self.time_step]])
```

But it doesn't seem to work with wp.sim.ModelBuilder(). I similarly used code to deform the mesh, but the mesh did not move.

```
    def simulate(self):
        wp.sim.collide(self.model, self.state_0)

        # self.model.shape_geo_src[0].vertices -> np.array
        # self.model.shape_geo_src[0].mesh.points -> <warp.types.array object at 0x7febf3d0ddb0>
        # self.animations[self.time_step] -> <warp.types.array object at 0x7febf3d0ddb0>
        wp.launch(
            kernel=deform, 
            dim=len(self.mesh.mesh.points), 
            inputs=[self.model.shape_geo_src[0].mesh.points, self.animations[self.time_step]]
        )

        for _ in range(self.sim_substeps):
            self.state_0.clear_forces()

            self.integrator.simulate(self.model, self.state_0, self.state_1, self.sim_dt)

            # swap states
            (self.state_0, self.state_1) = (self.state_1, self.state_0)
```
https://github.com/NVIDIA/warp/blob/7de917925ad67168497bcc7024e5f3b40ec84e56/warp/examples/sim/example_cloth.py#L130-L139

How can I deform mesh in wp.sim.ModelBuilder() ?
Should I add the mesh any other way than builder.add_shape_mesh ?

biansy000, 2024-07-12:
Same problem. I do not know how to add the ``deform`` function into the ``simulate`` function.

christophercrouzet, 2024-07-12:
Hi @SangHunHan92 and @biansy000,

`wp.sim.ModelBuilder()` is there to help setting up a simulation model. Once it's done, all the data is stored as Warp arrays in `wp.sim.Model` and `wp.sim.State`, so that's what you'd need to look at if you wanted to update anything while stepping through the simulation.

For example, to deform the mesh from the cloth example, you could run a kernel that'd modify `self.state_0.particle_q` before or after calling `self.simulate()`.

The example `warp/examples/sim/example_soft_body.py` is actually doing something along these lines when [launching the kernel `twist_points`](https://github.com/NVIDIA/warp/blob/69c5b8b72719f86d855d389630d1a746951a1240/warp/examples/sim/example_soft_body.py#L138-L142).