NVIDIA/warp #256: [BUG] from __future__ import annotations breaks warp

koen-v, 2024-07-05:
### Bug Description

When I add
`from __future__ import annotations`

to a warp script, the kernel wont compile, the first thing I hit is in:
python\warp\types.py 

`;'str' object has no attribute '__module__'`

removing the annotations import makes it go away

Thanks,
Koen

### System Information

Windows 11, Python 3.10

shi-eric, 2024-07-05:
A quick look at the issue shows that the type annotations get turned into strings with `from __future__ import annotations` ([postponed evaluation of annotations](https://peps.python.org/pep-0563/)), which interferes with the Warp `Var.type` variables created for the kernel parameters. Here's a minimal example :

```python
from __future__ import annotations

import warp as wp

wp.init()

@wp.kernel
def test_kernel(a: wp.array(dtype=wp.float32)):
    i = wp.tid()

test_array = wp.empty(shape=(10,), dtype=wp.float32)

wp.launch(test_kernel, test_array.shape, inputs=[test_array])

wp.synchronize()
```

@christophercrouzet: Can you look into if we can support this use case?

christophercrouzet, 2024-07-14:
Hi @koen-v, thanks for reporting this issue!

We've just merged some changes in [e10a583](https://github.com/NVIDIA/warp/commit/e10a583c38d87de49f4c9346f02cb9a89c7b7eb9) to add support for `from __future__ import annotations`. It's available in the branch `main` for now and will be part of the next release.

closed 2024-07-14 by christophercrouzet

koen-v, 2024-07-15:
great, thank you!