NVIDIA/warp #257: [BUG] '"cuda" device requested but this build of Warp does not support CUDA'

und-miller, 2024-07-08:
### Bug Description

When trying to run `python -m warp.examples.optim.example_drone` I get the following error below.

```
 File "C:\Users\user.name\Source\warp\warp\context.py", line 3265, in get_current_cuda_device
    raise RuntimeError('"cuda" device requested but this build of Warp does not support CUDA')
RuntimeError: "cuda" device requested but this build of Warp does not support CUDA
```

Warp installation details
```
PS C:\Users\user.name\Source\warp> pip show warp-lang
Name: warp-lang
Version: 1.2.2+cu12
Summary: A Python framework for high-performance simulation and graphics programming
Home-page:
Author:
Author-email: NVIDIA <mmacklin@nvidia.com>
License: NVIDIA Software License
Location: c:\users\user.name\appdata\local\programs\python\python39\lib\site-packages
Requires: numpy
Required-by:
```

CUDA Version: 12.4
Graphics Card: RTX 5000

How can I resolve this issue?

c0d1f1ed, 2024-07-08:
Thanks for reporting this! I installed this same version of Warp and it successfully runs this example and others on my RTX 3070 with CUDA 12.4 driver.

Did you install the GitHub release package or are you trying to build Warp from source?

und-miller, 2024-07-08:
Okay, so I did a complete fresh install of CUDA which resolved the CUDA error from above. However, I am getting a new error from the ShapeInstancer constructor when running `python -m warp.examples.optim.example_drone`. See below:

```File "C:\Users\stephen.alan.miller\Source\warp\warp\render\render_opengl.py", line 1273, in __init__
    self._axis_instancer = ShapeInstancer(self._shape_shader, self._device)
TypeError: __new__() takes 1 positional argument but 3 were given`
```

Do you want me to post this error in a different issue?

shi-eric, 2024-07-09:
Thanks for reporting @und-miller, I can reproduce the `ShapeInstancer` issue on my machine as well. I'll create an issue based on your comment for us to track.

shi-eric, 2024-07-09:
Closing this issue I believe the original issue is now solved.