NVIDIA/warp #266: Support of shared memory

jinz2014, 2024-07-16:
Can you please explain how to support shared memory in a kernel ?  Does the warp compiler optimize a kernel with shared memory ?  Thanks.

jinz2014, 2024-07-16:
For example, a dot product of two arrays. 

I suppose that the warp.dot() function computes a dot product of two vectors. Each vector is an element of an array.

daedalus5, 2024-07-17:
Hi @jinz2014 . Warp doesn't support shared memory in kernels directly, but you are free to use shared memory in native function snippets: https://nvidia.github.io/warp/modules/differentiability.html#custom-native-functions

jinz2014, 2024-07-17:
Hi @daedalus5 
I see. Will developers need to compute local ID (i.e. threadIdx.x) in a thread block ?  I think wp.tid() means global ID.

jinz2014, 2024-07-17:
Are there functions for local ID, thread block size, thread block ID ?

daedalus5, 2024-07-17:
Yes, `wp.tid()` is a global ID. We don't have functions in Python for those, but you should be able to access eg `threadIdx.x` in a native snippet as you would normally.

jinz2014, 2024-07-18:
Does snippet support template type ?     

snippet = 
'''
  __shared__ T sum[256]
'''

daedalus5, 2024-07-18:
No, I don't think templates would work in snippets currently.

toaster-robotics, 2024-08-24:
hello,
would anyone have code examples of how to use `func_native` to speed up **atomic add**? I've read that block-wise accumulation using shared memory, then using **atomic add** between blocks can speed things up vs just using `atomic_add`. However I'm unsure how to implement this using what's available in the [docs](https://nvidia.github.io/warp/modules/differentiability.html#custom-native-functions). I have the following questions after reading through that example code:

1. the example shows a 1d tid. But what do we do if tid is 2d or 3d?
2. the example has an input 1d array and an output int. how would you handle assigning the output to a slice of an array? for example I want to output to go into `A[i, j]`?

this is roughly what my code looks like

```python
@wp.kernel
def my_kernel(
    C_11: wp.array3d(dtype=wp.float32),  # type: ignore
    C_12: wp.array3d(dtype=wp.float32),  # type: ignore
    C_21: wp.array3d(dtype=wp.float32),  # type: ignore
    C_22: wp.array3d(dtype=wp.float32),  # type: ignore
) -> None:
    jj, kk = wp.tid()

    _C_11, _C_12, _C_21, _C_22 = calculations(jj, kk, ...)

    wp.atomic_add(C_11, jj, _C_11)
    wp.atomic_add(C_12, jj, _C_12)
    wp.atomic_add(C_21, jj, _C_21)
    wp.atomic_add(C_22, jj, _C_22)
```

closed 2024-10-11 by shi-eric