NVIDIA/warp #267: 'array' object does not support item assignment

jinz2014, 2024-07-17:
a = wp.empty(shape=10, dtype=int, device="cpu")
a[0] = 1

What is the recommended item assignment on a CPU ? Thanks.

jinz2014, 2024-07-17:
A related error is Item indexing is not supported on wp.array objects.

christophercrouzet, 2024-07-17:
Hi @jinz2014, this is by design. See the related entry in the FAQ: [Why aren’t assignments to Warp arrays supported outside of kernels?](https://nvidia.github.io/warp/faq.html#why-aren-t-assignments-to-warp-arrays-supported-outside-of-kernels).

christophercrouzet, 2024-07-17:
Note that you can call `arr.numpy()` to return the corresponding NumPy array on the CPU, and from there you'll be able to index/assign normally to it. If the `wp.array` data is already on the CPU, then this method will reuse the same data under the hood, without any copy, so it's cheap. However, if the `wp.array` is on the GPU, then `.numpy()` will copy the whole array to the CPU.

If you only need to read a single value, you can also use the slicing operator instead, like `arr[-1:].numpy()` for the last value.

jinz2014, 2024-07-17:
Thanks for the explanation! However, I don't understand the entry in the FAQ because they are array index/assign on a CPU. 

"For best performance, reading and writing data that is living on the GPU can only be performed inside Warp CUDA kernels. Otherwise individual element accesses such as array[i] = 1.0 in Python scope would require prohibitively slow device synchronization and copies."

I suppose there is no synchronization or copies for index/assign on a CPU when data lives on a CPU.

christophercrouzet, 2024-07-28:
That's right but the API exposed by `wp.array` needs to work for both GPU and CPU arrays. So if something isn't optimal in one case or the other, we don't expose it, if that makes sense?

jinz2014, 2024-07-29:
With the support of item assignment, users may replace numpy.array with warp.array for array initialization on a CPU. Is that right ?  I wonder if wp.array will replace numpy.array eventually.

christophercrouzet, 2024-07-29:
You can already initialize Warp arrays on the CPU by passing them a block of data at once, e.g.: `wp.array((1.0, 2.0, 3.0), dtype=wp.float32, device="cuda")`. You can also initialize a Warp array with a NumPy array. It's only individual item assignments that we'd like to not encourage since each assignment would require a kernel launch if the array lives on the GPU, and since we want to provide the same API for both CPU and GPU.

The goal of Warp arrays is to be compatible with NumPy arrays, but eventually their purpose is slightly different since the focus of Warp arrays is to have a unified way to store/process/read data regardless on where it lives.

closed 2024-08-26 by shi-eric