NVIDIA/warp #269: math functions for double-precision

jinz2014, 2024-07-19:
Are there math functions for double-precision floating-point operations ?  Thanks.

jinz2014, 2024-07-19:
I also got the error:

unsupported operand type(s) for *: 'float64' and 'float64'

shi-eric, 2024-07-19:
> Are there math functions for double-precision floating-point operations ? Thanks.

Yes, please take a look at the function signatures here to see if there is a double-precision version: https://nvidia.github.io/warp/modules/functions.html

> I also got the error:
> 
> unsupported operand type(s) for *: 'float64' and 'float64'
> 
> > > > a = wp.float64(0.1)
> > > > b = wp.float64(0.1)
> > > > a * b

I assume you're trying to do this at the Python scope, outside of a @wp.func or @wp.kernel decorated function? We don't currently define operator overloads for fundamental types like `wp.float64`. Try moving the code into the kernel scope. @christophercrouzet: Any thoughts here?

jinz2014, 2024-07-19:
There are no function for double-precision operations in Scalar Math. Right ?
 
Yes, at the Python scope.

christophercrouzet, 2024-07-25:
Hi @jinz2014, thanks for reporting this issue!

We've just merged the [commit 4094d6a](https://github.com/NVIDIA/warp/commit/4094d6a0347fb8339b95b37fc53628ce8dc6c8ee) in the branch `main` that addresses this, so it will be part of the next release!

closed 2024-07-25 by christophercrouzet