NVIDIA/warp #272: [BUG] Potential bug in warp.sim.collide?

xuan-li, 2024-07-19:
### Bug Description

In ```warp/sim/collide.py```, ```wp.clone``` are used to create new arrays when ```requires_grad=True```:

```
model.soft_contact_body_pos = wp.clone(model.soft_contact_body_pos)
model.soft_contact_body_vel = wp.clone(model.soft_contact_body_vel)
model.soft_contact_normal = wp.clone(model.soft_contact_normal)

...

model.rigid_contact_point0 = wp.clone(model.rigid_contact_point0)
model.rigid_contact_point1 = wp.clone(model.rigid_contact_point1)
model.rigid_contact_offset0 = wp.clone(model.rigid_contact_offset0)
model.rigid_contact_offset1 = wp.clone(model.rigid_contact_offset1)
model.rigid_contact_normal = wp.clone(model.rigid_contact_normal)
model.rigid_contact_thickness = wp.clone(model.rigid_contact_thickness)
```

It looks like that in the backpropagate, there will be a gradient flow between the new copy and the old one. But the two copies should be independent. I think ```wp.empty_like``` should be used here.

eric-heiden, 2024-07-19:
Hi @xuan-li,

Thanks for reporting this bug! You are right, the `wp.copy` operation now also copies the gradient arrays during the backward pass, so we should use `wp.empty_like` here. We will fix this for the next release.

etaoxing, 2025-01-22:
any updates on this? can submit a quick pr if needed.

shi-eric, 2025-01-22:
af57ae5 should contain a fix for the issue and will be included in the next Warp release at the beginning of February.

shi-eric, 2025-02-04:
Warp 1.6.0 is now out, so I'm making this issue as resolved.

closed 2025-02-04 by shi-eric