NVIDIA/warp #273: [QUESTION] License in plain language -- does it allow for non-commercial derivative works?

maria-korosteleva, 2024-07-21:
Hi! Thank you so much for sharing and developing such an amazing library!

I've used warp extensively in my recent academic research project, although we had to add some routines (e.g. additional collision handling kernels and constraints kernels to the XPBD simulator). Since the start of our project, the Warp license has been updated (since v 0.13.0), and now I cannot fully understand the terms of the license and how it handles derivative works

E.g. can we share our fork of Warp with the abovementioned changes here on GitHub? Of course, it will preserve the licensing notes in the code, and the text of the license

It may come to a more general question -- does the license allow GitHub forks in general? 

I appreciate any help you can provide

mmacklin, 2024-07-22:
Hi @maria-korosteleva, as you noticed the license was updated recently to be more permissive regarding derived works and commercial use. Forking the project is fine and should be covered by Section 1.2.

Your changes sound useful though so if you believe we should integrate them back to the main Warp branch then please open a PR!

Cheers,
Miles

maria-korosteleva, 2024-07-22:
That's great! Thank you for the clarification =)

I will try to make it happen =) I will first release it as a fork, and then
see if I have the time to refactor it into something that could be
integrated

Best,
Maria

closed 2024-10-11 by shi-eric