NVIDIA/warp #282: [REQUEST] C and Python Modulus operator for vectorial quantities

fritzio, 2024-07-29:
### Description

- The modulus operator `mod` is currently only defined for scalar, i.e., `float` quantities. I suggest implementing it for vectorial and tensorial quantities, e.g., `vec3`, so that we can omit a wrapper function, such as,
```
@wp.func
def mod(a: wp.vec3, b: wp.vec3) -> wp.vec3:
    return wp.vec3(wp.mod(a[0], b[0]), wp.mod(a[1], b[1]), wp.mod(a[2], b[2]))
```

- The modulus operator `mod` gives the modulus as defined in `C` and not in `Python`, which might lead to surprising behavior if we are unaware. I suggest adding a Pythonic modulus operator, e.g.,
```
@wp.func
def pymod(a: float, b: float) -> float:
    return wp.mod(wp.mod(a, b) + b, b)
```

### Context
The motivation for this request is to have less boilerplate code in our application.

sudo-panda, 2024-08-18:
Can I take this up?

shi-eric, 2024-08-18:
I think @christophercrouzet is actively working on this issue already.

christophercrouzet, 2024-08-18:
We just merged the changes to the branch `main`. So far we only added support for vectors to `wp.mod()` and documented the built-in as being implemented as a truncated division to alleviate the ambiguity with Python's implementation. We haven't implemented the floored division variant for now.

closed 2024-08-18 by christophercrouzet