NVIDIA/warp #284: [REQUEST] Add atomic_max and atomic_min for float64

swan2818, 2024-07-31:
### Description
For now, the atomic_max and atomic_min operators in Warp are for float and int only.

### Context
Please support float64 for the atomic_max and atomic_min operators.

Irene-123, 2024-08-11:
Hi @swan2818 @shi-eric 
I went through this issue and looking to contribute here

shi-eric, 2024-08-11:
Hi @Irene-123, thanks for volunteering! I think @daedalus5 might already be working on this issue. The other two you expressed an interest in should be open for you to take up.

Irene-123, 2024-08-11:
That's great, thank you @shi-eric 
Is there a discord channel to have a discussion?

shi-eric, 2024-08-11:
> That's great, thank you @shi-eric Is there a discord channel to have a discussion?

Yes, there is a channel for Warp in the Omniverse Discord server ([link](https://discord.com/invite/nvidiaomniverse))

daedalus5, 2024-08-12:
Yes I have an MR in for this now.

closed 2024-10-01 by shi-eric