NVIDIA/warp #288: [BUG] bvh_query_ray only returns an intersection when query_start is within the aabb

jnpmartel, 2024-08-11:
### Bug Description

Dear NVIDIA warp team, 

First let me say NVIDIAwarp is a fantastic project: I have been using it a lot for prototyping various simulation and rendering projects. 

I believe there is a bug in the way the bvh query works. This is the simplest snippet of code I could write to reproduce it:
- It creates num_bounds aabbs that span [0.5,1.0] on the x axis (they are all on top of each others in this example)
- It creates a ray in the positive direction along the x axis
The expected output is to see a print "intersected bounds=<num_bounds>" (=9 in this example)

**Bug:** This piece of code returns "intersected bounds=9" only when the query_start (the origin of the ray) is contained within the aabbs, for instance query_start = wp.vec3(0.75, 0.0, 0.0) but returns "intersected bounds=0" otherwise (for instance query_start = wp.vec3(0.0, 0.0, 0.0), the ray is in front of the boxes) 

```python
import warp as wp
import numpy as np

@wp.kernel
def bvh_query_ray(
    bvh_id: wp.uint64,
    ray_org: wp.vec3,
    ray_dir: wp.vec3,
    bounds_intersected: wp.array(dtype=wp.bool),
):
    query = wp.bvh_query_ray(bvh_id, ray_org, ray_dir)
    bounds_nr = wp.int32(0)

    while wp.bvh_query_next(query, bounds_nr):
        # The ray intersects the volume with index bounds_nr
        bounds_intersected[bounds_nr] = True

num_bounds = 9

lowers = np.tile(np.array([0.5,-1.,-1.])[np.newaxis,:],(num_bounds,1))
uppers = np.tile(np.array([1.0, 1., 1.])[np.newaxis,:],(num_bounds,1))

device_lowers = wp.array(lowers, dtype=wp.vec3f, device="cuda:0")
device_uppers = wp.array(uppers, dtype=wp.vec3f, device="cuda:0")
bounds_intersected = wp.zeros(shape=(num_bounds), dtype=wp.bool, device="cuda:0")

bvh = wp.Bvh(device_lowers, device_uppers)
bvh.refit()

query_start = wp.vec3(0.75, 0.0, 0.0)
query_dir = wp.normalize(wp.vec3(1.0, 0.0, 0.0))

wp.launch(
    kernel=bvh_query_ray,
    dim=1,
    inputs=[bvh.id, query_start, query_dir, bounds_intersected],
    device="cuda:0",
)

bounds = wp.to_torch(bounds_intersected)
print(f"intersected bounds={bounds.sum()}")
```

jnpmartel, 2024-08-11:
Also looking at the BVH [unit-tests](https://github.com/NVIDIA/warp/blob/07d4f5bbe93928950cc8830e5cdc0488b6447f0e/warp/tests/test_bvh.py#L26) that you have, it is not clear you would catch the bug since it seems the unit test always has ray_start within the AABBs

jnpmartel, 2024-08-11:
Actually, I am pretty confused:

[this implementation of intersect_aabb](https://github.com/NVIDIA/warp/blob/main/warp/native/intersect.h#L133) seems pretty much correct to me (contrary to the one I mentionned aboved in your unit tests) if (and only if) you were to provide the reciprocal direction (1./dir) as a second argument the function (as the name suggests). This would help handle the nagative direction cases. 

However, from what I read in the [bvh_query implementation](https://github.com/NVIDIA/warp/blob/07d4f5bbe93928950cc8830e5cdc0488b6447f0e/warp/native/bvh.h#L242) you provide it with the actual direction (not its inverse).

mmacklin, 2024-08-15:
Hi @jnpmartel, I think you're right this looks like a clear bug that `bvh_query()` should be passing the reciprocal dir as the `upper` argument.

@christophercrouzet could you please take a look at this?

christophercrouzet, 2024-08-18:
Thanks a lot @jnpmartel for raising this issue and also resolving it! We've just merged the fix into the branch `main` and made sure to add a regression test using your repro.

closed 2024-08-18 by christophercrouzet

yzx9, 2024-08-29:
@shi-eric Hi, I hope this message finds you well. I noticed a small mistake: the fix commit message should be 'Fix a bug in `wp.bvh_query_ray()`'. This correction should also be reflected in the changelog.

christophercrouzet, 2024-08-29:
My bad, thanks for pointing this out @yzx9! The changelog entry will be updated.

shi-eric, 2024-08-30:
This issue should be fixed with the 1.3.2 release: https://github.com/NVIDIA/warp/releases