NVIDIA/warp #293: [BUG] Calling bsr_set_from_triplets for multiple BSR matrices gives NaN

Yidong-ZHAO, 2024-08-19:
### Bug Description

### Bug description
Calling `bsr_set_from_triplets` for multiple BSR matrices leads to a NaN error.

### To reproduce
```python
import warp as wp
import warp.sparse 

rows = wp.zeros(4, dtype=wp.int32)
cols = wp.zeros(4, dtype=wp.int32)
vals = wp.zeros(4, dtype=wp.float64)

bsr_matrix1 = wp.sparse.bsr_zeros(2, 2, block_type=wp.float64)
bsr_matrix2 = wp.sparse.bsr_zeros(2, 2, block_type=wp.float64)

wp.sparse.bsr_set_from_triplets(bsr_matrix1, rows, cols, vals)
print('bsr_matrix1:', bsr_matrix1)

wp.sparse.bsr_set_from_triplets(bsr_matrix2, rows, cols, vals)
print('bsr_matrix2:', bsr_matrix2)
```

### Results
`bsr_matrix2` contains NaNs.

### System Information

Warp version: 1.3.1
CUDA version: 12.5
OS: Ubuntu 22.04
Python version: 3.12.4

gdaviet, 2024-08-19:
Hi @Yidong-ZHAO, what you are seeing is uninitialized memory. By default `bsr_from_triplets` will get rid of null blocks, so the matrix is actually empty (this can be changed by passing `prune_numerical_zeros=False` to `bsr_from_triplets`).

Now the confusing bit may be the `nnz=4` that you see when printing the matrices. But this is actually an upper bound, that is used for allocating memory and dimensioning kernel launches, not the exact number of non-zeros: the exact number is given by `matrix.offsets[matrix.nrow-1]`, in device memory. To avoid forcing a synchronization (which would prevent capture in CUDA graph), warp.sparse does not automatically transfer the exact nnz to the host, however it provides a `matrix.nnz_sync()` function that you may call perform this synchronization. In your example, you will see that `nnz_sync()` returns `0` for both matrices.

So in short:
 - The only coefficients in `columns` and `values` that will actually get used those in the range (0, matrix.nnz_sync())
 - If you don;t want `bsr_from_triplets` to remove the zero coefficients, pass the  `prune_numerical_zeros=False`  flag

Yidong-ZHAO, 2024-08-20:
Hi @gdaviet, thank you for your detailed explanation. Now I understand.

Since the question has been answered, I will close the issue.

closed 2024-08-20 by Yidong-ZHAO