NVIDIA/warp #300: [QUESTION] <title>Error: Could not open libcuda.so.

MarcYu0303, 2024-08-28:
A problem of running the warp, here is the detailed information:

```
opencv-contrib not installed, some features will be disabled.
Please install with `pip3 install opencv-contrib-python`
Error: Could not open libcuda.so.
Warp initialized:
   Version: 0.3.1
   CUDA device: NVIDIA RTX A6000
   Kernel cache: /home/yuran/.cache/warp/0.3.1
2024-08-28 15:25:23,037 - mani_skill2 - INFO - load sdfs from file
Traceback (most recent call last):
  File "src/test_maniskill_env.py", line 7, in <module>
    env = gym.make("Pinch-v0", obs_mode="rgbd", control_mode="pd_joint_delta_pos")
  File "/home/yuran/anaconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "/home/yuran/anaconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 129, in make
    env = spec.make(**kwargs)
  File "/home/yuran/anaconda3/envs/mani_skill2/lib/python3.8/site-packages/gym/envs/registration.py", line 90, in make
    env = cls(**_kwargs)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/mani_skill2/envs/mpm/pinch_env.py", line 36, in __init__
    super().__init__(*args, **kwargs)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/mani_skill2/envs/mpm/base_env.py", line 95, in __init__
    super().__init__(*args, **kwargs)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/mani_skill2/envs/sapien_env.py", line 132, in __init__
    obs = self.reset(reconfigure=True)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/mani_skill2/envs/mpm/pinch_env.py", line 46, in reset
    return super().reset(*args, seed=seed, **kwargs)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/mani_skill2/envs/mpm/base_env.py", line 297, in reset
    return super().reset(*args, **kwargs)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/mani_skill2/envs/sapien_env.py", line 402, in reset
    self.reconfigure()
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/mani_skill2/envs/mpm/base_env.py", line 111, in reconfigure
    self._setup_mpm()
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/mani_skill2/envs/mpm/pinch_env.py", line 56, in _setup_mpm
    self.mpm_model = self.model_builder.finalize(device="cuda")
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/warp_maniskill/mpm/mpm_model.py", line 458, in finalize
    m0: MPMModel = super().finalize(device)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/warp_maniskill/warp/sim/model.py", line 1872, in finalize
    m.particle_q = wp.array(self.particle_q, dtype=wp.vec3, device=device)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/warp_maniskill/warp/types.py", line 534, in __init__
    dest = empty(shape, dtype=dtype, device=device, requires_grad=requires_grad)
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/warp_maniskill/warp/context.py", line 1056, in empty
    return zeros(shape, dtype, device, requires_grad=requires_grad, **kwargs)  
  File "/home/yuran/Projects/Mani_Derfom_GS3d/ManiSkill2-docs/warp_maniskill/warp/context.py", line 984, in zeros
    raise RuntimeError("Trying to allocate CUDA buffer without GPU support")
RuntimeError: Trying to allocate CUDA buffer without GPU support

```

does anyone meet the similar problem, is there any suggestion about this? Thanks a lot!

shi-eric, 2024-08-28:
Hey @MarcYu0303, this is quite an old version of Warp which could make it difficult for us to diagnose. Is it possible to try running the latest version of Warp in a different Conda environment? It doesn't have to be this `warp_maniskill` problem, maybe just see if you can allocate an array on the GPU.

shi-eric, 2024-08-28:
Can you also clarify if this is WSL2?

MarcYu0303, 2024-08-29:
Thanks for your reply. I find the similar problem when running some other simulator, which means this problem is not related to Warp itself. And finally I solved it. Thanks again for your time!

closed 2024-08-29 by MarcYu0303