NVIDIA/warp #305: [BUG] "bvh_query_xxxx" is not a known attribute of module "warp"

yzx9, 2024-09-03:
### Bug Description

The BVH runtime functions `bvh_query_aabb`, `bvh_query_ray`, and `bvh_query_next` are [documented](https://nvidia.github.io/warp/modules/functions.html#warp.bvh_query_aabb) and work correctly at runtime, but they are not typed in the library.

I noticed that these functions are missing from the `stub.py` file, while the `volume_xxxxx` functions are present. This suggests that something may have gone wrong during the generation of `stub.py`.

## Steps to Reproduce

1. Open a file and type the following:

   ```python
   import warp
   warp.bvh_query_aabb()
   ```

2. You will receive an error from Pyright: `"bvh_query_aabb" is not a known attribute of module "warp" 152:16:10 Pyright reportAttributeAccessIssue`.

### System Information

```
Warp 1.3.1 initialized:
   CUDA devices not available
   Devices:
     "cpu"      : "CPU"
   Kernel cache:
     /home/yzx9/.cache/warp/1.3.1
```

christophercrouzet, 2024-09-03:
Hi @yzx9, thanks for raising this issue! You're correct and last week we actually opened a MR internally that addresses the built-ins missing from `stubs.py`. I'll comment here once it's merged!

christophercrouzet, 2024-09-05:
Hi @yzx9, the commit [7847af9](https://github.com/NVIDIA/warp/commit/7847af9ac74be12bc0cb90efd8cd1ff6d271b5a2) has now been merged to `main` and will be part of a future release. Apologies for the inconvenience caused!

closed 2024-09-05 by christophercrouzet