NVIDIA/warp #309: Regarding running basic example_render_opengl.py

mmmovania, 2024-09-08:
Hi all,  
   Thanks for sharing this library. I have downloaded the latest version of warp (master branch). I am trying to run the example_render_opengl.py on my Windows 10 x64 PC with CUDA v11.6. I have done the warp setup from src and all warp tests succeeded. I then ran the command that was written on the readme page that is 

`python -m warp.examples.core.example_render_opengl.py`

But this fails and gives me the following error.

`C:\python38\python.exe: Error while finding module specification for 'warp.examples.core.example_render_opengl.py' (ModuleNotFoundError: __path__ attribute not found on 'warp.examples.core.example_render_opengl' while trying to find 'warp.examples.core.example_render_opengl.py')`

When I opened the example_render_opengl.py in VSCode to run it from there, I get this error.
Exception has occurred: AssertionError
```
Warp not initialized, call wp.init() before use
  File "E:\warp-main\warp\examples\core\example_render_opengl.py", line 99, in <module>
    with wp.ScopedDevice(args.device):
AssertionError: Warp not initialized, call wp.init() before use
```
I tried to find this error in the issues before posting but there seems to be nothing on this. How can i solve this issue?

Thanks,
M. Mobeen

shi-eric, 2024-09-08:
I think the first command needs to omit the `.py`: `python -m warp.examples.core.example_render_opengl`

For the second issue, try inserting a `wp.init()` near the top of the example. Is there anything else in the logs from the case without a `wp.init()` besides the error you shared?

We changed the code recently not to require an explicit `wp.init()` except in rare cases, so something was probably overlooked by us here.

mmmovania, 2024-09-08:
Thanks for a prompt reply @shi-eric yes indeed the .py is not needed and it works fine.
> I think the first command needs to omit the `.py`: `python -m warp.examples.core.example_render_opengl`

> For the second issue, try inserting a `wp.init()` near the top of the example. Is there anything else in the logs from the case without a `wp.init()` besides the error you shared?
Yeah I added wp.init() before the with wp.ScopedDevice(args.device) and it works fine. thanks. Here is the output that i see in my VSCode. Both issues are resolved.

[image]

closed 2024-09-08 by mmmovania