NVIDIA/warp #311: [BUG] Incorrect `iter_reverse` function for range

fbxiang, 2024-09-10:
### Bug Description

The `iter_reverse` function for range seems to only work for `step = 1`.
https://github.com/NVIDIA/warp/blob/3b00f16a3e2a96415625eeb22139c8476e4170fe/warp/native/range.h#L96-L107
I think it needs to be something like the following? (Not thoroughly tested)
```c
    rev.start = r.start + (r.end - r.start - sign(r.step)) / r.step * r.step;
    rev.end = rev.start - r.end + r.start;
```

christophercrouzet, 2024-10-01:
Hi @fbxiang, thank you for reporting this issue and apologies for any inconvenience caused!

Your proposed fix is correct and we've prepared a MR internally to address it.

christophercrouzet, 2024-10-13:
It's been merged as part of [commit c0dbdf9](https://github.com/NVIDIA/warp/commit/c0dbdf9b09b7879b4a1e5a045f4c7448fce6348e) and will be part of the next release of Warp, thanks again!

closed 2024-10-13 by christophercrouzet