NVIDIA/warp #317: [BUG] Garbage value in wp.array

sin3point14, 2024-09-26:
### Bug Description

Running
```py
import warp as wp

@wp.kernel
def ker(hessians: wp.array(dtype = wp.float32, ndim = 3)):
    t = wp.tid()
    for i in range(6):
        for j in range(6):
            hessians[t][i][j] = 1.0

    wp.printf("hessians %d\n[0, 0] %f [0, 1] %f [0, 2] %f [0, 3] %f [0, 4] %f [0, 5] %f\n [1, 0] %f [1, 1] %f [1, 2] %f [1, 3] %f [1, 4] %f [1, 5] %f\n [2, 0] %f [2, 1] %f [2, 2] %f [2, 3] %f [2, 4] %f [2, 5] %f\n [3, 0] %f [3, 1] %f [3, 2] %f [3, 3] %f [3, 4] %f [3, 5] %f\n [4, 0] %f [4, 1] %f [4, 2] %f [4, 3] %f [4, 4] %f [4, 5] %f\n [5, 0] %f [5, 1] %f [5, 2] %f [5, 3] %f [5, 4] %f [5, 5] %f\n", t, hessians[t][0][0], hessians[t][0][1], hessians[t][0][2], hessians[t][0][3], hessians[t][0][4], hessians[t][0][5], hessians[t][1][0], hessians[t][1][1], hessians[t][1][2], hessians[t][1][3], hessians[t][1][4], hessians[t][1][5], hessians[t][2][0], hessians[t][2][1], hessians[t][2][2], hessians[t][2][3], hessians[t][2][4], hessians[t][2][5], hessians[t][3][0], hessians[t][3][1], hessians[t][3][2], hessians[t][3][3], hessians[t][3][4], hessians[t][3][5], hessians[t][4][0], hessians[t][4][1], hessians[t][4][2], hessians[t][4][3], hessians[t][4][4], hessians[t][4][5], hessians[t][5][0], hessians[t][5][1], hessians[t][5][2], hessians[t][5][3], hessians[t][5][4], hessians[t][5][5])

wp.config.verify_cuda = True
dim = 1
hessians = wp.empty((dim, 6, 6), dtype = wp.float32)
hessians.zero_()

wp.launch(kernel=ker,
        dim=dim,
        inputs=[hessians])
```

I expect to see all values as 1.0, but...

```
Warp 1.3.3 initialized:
   CUDA Toolkit 12.5, Driver 12.5
   Devices:
     "cpu"      : "Intel64 Family 6 Model 154 Stepping 3, GenuineIntel"
     "cuda:0"   : "NVIDIA GeForce RTX 3070 Ti Laptop GPU" (8 GiB, sm_86, mempool enabled)
   Kernel cache:
     C:\Users\manas\AppData\Local\NVIDIA\warp\Cache\1.3.3
Module __main__ 944968a load on device 'cuda:0' took 478.31 ms  (compiled)
hessians 0
[0, 0] 1.000000 [0, 1] 1.000000 [0, 2] 1.000000 [0, 3] 1.000000 [0, 4] 1.000000 [0, 5] 1.000000
 [1, 0] 1.000000 [1, 1] 1.000000 [1, 2] 1.000000 [1, 3] 1.000000 [1, 4] 1.000000 [1, 5] 1.000000
 [2, 0] 1.000000 [2, 1] 1.000000 [2, 2] 1.000000 [2, 3] 1.000000 [2, 4] 1.000000 [2, 5] 1.000000
 [3, 0] 1.000000 [3, 1] 1.000000 [3, 2] 1.000000 [3, 3] 1.000000 [3, 4] 1.000000 [3, 5] 1.000000
 [4, 0] 1.000000 [4, 1] 1.000000 [4, 2] 1.000000 [4, 3] 1.000000 [4, 4] 1.000000 [4, 5] 1.000000
 [5, 0] 1.000000 [5, 1] 106209064793978645298368182800580483424074505329040650509470198072093055602238913718558306764935615593345335844991279653254512959868664681658128284707651926600821908889183478784911917359053337781112180476656894676903916948987801880858781155386720256.000000 [5, 2] 0.000000 [5, 3] 0.000000 [5, 4] 0.000000 [5, 5] 0.000000
```
Shifting to CPU device, fixes it
```diff
diff --git a/isolated.py b/isolated.py
index 0e6455c..27e1590 100644
--- a/isolated.py
+++ b/isolated.py
@@ -10,6 +10,7 @@ def ker(hessians: wp.array(dtype = wp.float32, ndim = 3)):
     wp.printf("hessians %d\n[0, 0] %f [0, 1] %f [0, 2] %f [0, 3] %f [0, 4] %f [0, 5] %f\n [1, 0] %f [1, 1] %f [1, 2] %f [1, 3] %f [1, 4] %f [1, 5] %f\n [2, 0] %f [2, 1] %f [2, 2] %f [2, 3] %f [2, 4] %f [2, 5] %f\n [3, 0] %f [3, 1] %f [3, 2] %f [3, 3] %f [3, 4] %f [3, 5] %f\n [4, 0] %f [4, 1] %f [4, 2] %f [4, 3] %f [4, 4] %f [4, 5] %f\n [5, 0] %f [5, 1] %f [5, 2] %f [5, 3] %f [5, 4] %f [5, 5] %f\n", t, hessians[t][0][0], hessians[t][0][1], hessians[t][0][2], hessians[t][0][3], hessians[t][0][4], hessians[t][0][5], hessians[t][1][0], hessians[t][1][1], hessians[t][1][2], hessians[t][1][3], hessians[t][1][4], hessians[t][1][5], hessians[t][2][0], hessians[t][2][1], hessians[t][2][2], hessians[t][2][3], hessians[t][2][4], hessians[t][2][5], hessians[t][3][0], hessians[t][3][1], hessians[t][3][2], hessians[t][3][3], hessians[t][3][4], hessians[t][3][5], hessians[t][4][0], hessians[t][4][1], hessians[t][4][2], hessians[t][4][3], hessians[t][4][4], hessians[t][4][5], hessians[t][5][0], hessians[t][5][1], hessians[t][5][2], hessians[t][5][3], hessians[t][5][4], hessians[t][5][5])

 wp.config.verify_cuda = True
+wp.set_device("cpu")
 dim = 1
 hessians = wp.empty((dim, 6, 6), dtype = wp.float32)
 hessians.zero_()
```
output
```
Warp 1.3.3 initialized:
   CUDA Toolkit 12.5, Driver 12.5
   Devices:
     "cpu"      : "Intel64 Family 6 Model 154 Stepping 3, GenuineIntel"
     "cuda:0"   : "NVIDIA GeForce RTX 3070 Ti Laptop GPU" (8 GiB, sm_86, mempool enabled)
   Kernel cache:
     C:\Users\manas\AppData\Local\NVIDIA\warp\Cache\1.3.3
Module __main__ 944968a load on device 'cpu' took 1.93 ms  (cached)
hessians 0
[... 33 lines cut ...]
   Kernel cache:
     C:\Users\manas\AppData\Local\NVIDIA\warp\Cache\1.3.3
Module __main__ 6ac2b41 load on device 'cuda:0' took 468.73 ms  (compiled)
hessians 0
[0, 0] 1.000000 [0, 1] 1.000000 [0, 2] 1.000000 [0, 3] 1.000000 [0, 4] 1.000000 [0, 5] 1.000000
 [1, 0] 1.000000 [1, 1] 1.000000 [1, 2] 1.000000 [1, 3] 1.000000 [1, 4] 1.000000 [1, 5] 1.000000
 [2, 0] 1.000000 [2, 1] 1.000000 [2, 2] 1.000000 [2, 3] 1.000000 [2, 4] 1.000000 [2, 5] 1.000000
 [3, 0] 1.000000 [3, 1] 1.000000 [3, 2] 1.000000 [3, 3] 1.000000 [3, 4] 1.000000 [3, 5] 1.000000
 [4, 0] 1.000000 [4, 1] 1.000000 [4, 2] 1.000000 [4, 3] 1.000000 [4, 4] 1.000000 [4, 5] 1.000000
 [5, 0] 1.000000 [5, 1] 73736568 [5, 2] 0.000000 [5, 3] 0.000000 [5, 4] 0.000000 [5, 5] 0.000000
```

I had this `hessian` as part of a super long kernel and even in that case as well `[5, 1]` was being set to this weird hex value and `[5, 2]` -> `[5, 5]` were 0 everytime

### System Information

```
PS C:\Users\manas\my-projs> pip freeze | grep warp
warp-lang==1.3.3
PS C:\Users\manas\my-projs> python --version
Python 3.12.2
PS C:\Users\manas\my-projs> cmd /c ver

Microsoft Windows [Version 10.0.26120.1843]
```

sin3point14, 2024-09-26:
I was trying to test this on colab and for some reason I can't see the output of `wp.printf`:
[image]
Can we have some documentation on how to print on colab? Or maybe this too is some bug?

christophercrouzet, 2024-10-02:
Hi @sin3point14,

The issue you're running into is happening due to the following CUDA limitation documented [here](https://docs.nvidia.com/cuda/cuda-c-programming-guide/index.html#limitations):

> The `printf()` command can accept at most 32 arguments in addition to the format string. Additional arguments beyond this will be ignored, and the format specifier output as-is.

Indeed, the following snippet fails for the 33rd entry:

```py
#!/usr/bin/env python3

import warp as wp

@wp.kernel
def foo():
    wp.printf(
        " 0=%d,  1=%d,  2=%d,  3=%d,  4=%d,  5=%d,  6=%d,  7=%d, "
        " 8=%d,  9=%d, 10=%d, 11=%d, 12=%d, 13=%d, 14=%d, 15=%d, "
        "16=%d, 17=%d, 18=%d, 19=%d, 20=%d, 21=%d, 22=%d, 23=%d, "
        "24=%d, 25=%d, 26=%d, 27=%d, 28=%d, 29=%d, 30=%d, 31=%d, "
        "32=%d\n",
         0,  1,  2,  3,  4,  5,  6,  7,
         8,  9, 10, 11, 12, 13, 14, 15,
        16, 17, 18, 19, 20, 21, 22, 23,
        24, 25, 26, 27, 28, 29, 30, 31,
        32,
    )

def run():
    wp.launch(kernel=foo, dim=1)
    wp.synchronize()

if __name__ == "__main__":
    with wp.ScopedDevice("cuda"):
        run()
```

As a result, the usage of `wp.printf()` should probably be restricted to debugging a few variables here and there but, if you'd like to print a whole array, then logging this outside of a kernel, as you've done it with `print(hessians.numpy())`, sounds like a more robust approach.

christophercrouzet, 2024-10-02:
As for `wp.printf()` not showing up, you might need to call `wp.synchronize()` after launching the kernel.

closed 2024-10-02 by christophercrouzet

sin3point14, 2024-10-03:
Maybe some warning can be printed for this? Or maybe it did and i missed it