NVIDIA/warp #324: [BUG] Marching Cubes off by 1

eschneeweiss, 2024-10-07:
### Bug Description

I was replacing my marching cubes code that uses skimage with warp's implementation to speed things up, but I think I discovered an issue. It seems like the outer bounds of marching cubes is being ignored. To test this if you create a volume of all -1s and then set the inside to 1s creating a volume of 1s with a boundary of -1, the result of marching cubes will not produce a cube, only 3 sides of the cube will be created (the sides closest to 0). Additionally if you make the -1 boundary 2 thick this resolves the issue.

I think I know where the issue is but I have not tested it, in marching.cu create_cell_tris line 318:
```
        if (xi >= mc.nx - 2 || yi >= mc.ny - 2 || zi >= mc.nz - 2)
            return;
```
I think it should be either >= or -2 but not both, for example: ```xi > mc.nx -2``` or ```xi >= mc.nx -1```

I have attached two images to show what warp's marching cubes looks like compared to that of skimage.

[image]
[image]

christophercrouzet, 2024-10-10:
Hi @eschneeweiss, thank you for opening this issue and apologies for any inconvenience caused!

We actually became aware of this bug some time ago and did experiment with what you suggested, however it wasn't enough to fix the issue that we found with our test geometry, and we didn't spend any more time on it since it hasn't been much of a priority so far.

We will update this issue once it's prioritized and addressed, thanks again!

eschneeweiss, 2024-10-11:
Hi @christophercrouzet thank you for getting back to me. What can I do to help prioritize this? XD

christophercrouzet, 2024-10-15:
We were hoping to look into it by the end of the year, or early next year. Is it critical on your end?

eschneeweiss, 2024-10-15:
> We were hoping to look into it by the end of the year, or early next year. Is it critical on your end?

I guess it is not that urgent. Thank you for looking into this!

closed 2025-07-18 by shi-eric