NVIDIA/warp #329: [BUG] Hard-coded collision value will cause small deformable bodies to be unstable/inaccurate.

mmichelis, 2024-10-16:
### Bug Description

In warp/sim/integrator_euler.py line 306 there are hard-coded collision checks between triangles that are less than 0.01 apart. When loading a soft deformable mesh that is on millimeter scale, these collision checks will come into play and cause artificial damping in best case, and divergence in worst case scenarios. The symplectic integrator will no longer be energy-conserving when this self-collision check is done between each particle in the soft mesh.

AnkaChan, 2024-10-16:
Hi @mmichelis, thank you for bringing this up. I will look into it. We can make this value an adjustable parameter, such as `soft_contact_radius` within `Model`.

mmichelis, 2024-10-17:
Excellent, thanks!

mmichelis, 2024-10-17:
Excellent, thanks!

shi-eric, 2024-10-17:
Let's leave the issue open until the fix is in so we can keep an eye on it.

shi-eric, 2024-12-30:
Looks like this was fixed in eacad4f827a1e0d6e11e66c6819f398d1425c9fa and released in Warp 1.5.0.

closed 2024-12-30 by shi-eric