NVIDIA/warp #330: Custom additions to Model and Custom Controls?

Sina-Haz, 2024-10-21:
Hi, I was wondering if there is some kind of interface so that I can define a struct or some class and have it be an object in Warp's model class? I want to implement cables attached to rigid bodies and have as a custom control the changing of the rest lengths.

eric-heiden, 2024-10-27:
Hi @Sina-Haz, we don't have such an interface yet where custom elements can be added to the model or control object. But you could simply assign your own structures to the `Model` or `Control` objects that you need for cable simulation and perform some custom handling before/after `Integrator.simulate()` yourself.

Sina-Haz, 2024-10-28:
I see. Could you give me an example of this being done. Doesn't necessarily have to be my use case but any example code would be much appreciated.

eric-heiden, 2024-12-17:
You can take a look at the drone example where we add a custom field `prop_controls` to the `wp.sim.Control` object:

https://github.com/NVIDIA/warp/blob/main/warp/examples/optim/example_drone.py#L418