NVIDIA/warp #334: [BUG] Inconsistent behaviour for when creating arrays from numpy arrays on multiple gpus

elrnv, 2024-10-25:
### Bug Description

Creating warp arrays from numpy arrays seems to fail when run on secondary gpu.

MVP:
```python
import numpy as np
import warp as wp

dev = "cuda:1"
a = np.array([1])
b = np.array([2])
wp_a = wp.from_numpy(a).to(dev)
wp_b = wp.from_numpy(b).to(dev)
tc_a = wp.to_torch(wp_a)
tc_b = wp.to_torch(wp_b)
print(a, b)
print(wp_a, wp_b)
a = np.array([3])
b = np.array([4])
wp_a = wp.from_numpy(a).to(dev)
wp_b = wp.from_numpy(b).to(dev)
tc_a = wp.to_torch(wp_a)
tc_b = wp.to_torch(wp_b)
print(a, b)
print(wp_a, wp_b)
assert np.all(tc_a.cpu().numpy() == a)
assert np.all(tc_b.cpu().numpy() == b)
```

expected result:
```
[1] [2]
[1] [2]
[3] [4]
[3] [4]
```
and no assertion.

Actual result:
```
[1] [2]
[1] [2]
[3] [4]
[2] [4]
Traceback (most recent call last):
  File "C:\open\proj\hermes\warp_fail.py", line 22, in <module>
    assert np.all(tc_a.cpu().numpy() == a)
AssertionError
```

### System Information

Warp initialization info:

```
Warp 1.4.0 initialized:
   CUDA Toolkit 12.6, Driver 12.2
   Devices:
     "cpu"      : "AMD64 Family 23 Model 49 Stepping 0, AuthenticAMD"
     "cuda:0"   : "NVIDIA RTX A6000" (48 GiB, sm_86, mempool enabled)
     "cuda:1"   : "NVIDIA RTX A6000" (48 GiB, sm_86, mempool enabled)
   CUDA peer access:
     Not supported
   Kernel cache:
     C:\Users\elrnv\AppData\Local\NVIDIA\warp\Cache\1.4.0
```

nvlukasz, 2024-10-29:
Hi @elrnv, there's a synchronization issue with this code.  To fix it, you can modify these lines:
```python
wp_a = wp.from_numpy(a).to(dev)
wp_b = wp.from_numpy(b).to(dev)
```

Like this:
```python
wp_a = wp.from_numpy(a, device=dev)
wp_b = wp.from_numpy(b, device=dev)
```

This should also improve the performance.

Let me explain.  The `wp.from_numpy(a)` call creates the Warp array on the default device, which is "cuda:0".  This triggers a CUDA memory copy from the host to "cuda:0", which runs in the context of "cuda:0".  The `.to(dev)` call launches a CUDA memory copy from "cuda:0" to "cuda:1".  By default memory copies between different GPUs run in the context of the destination device, which is "cuda:1" in this case.  Since operations on different devices are not synchronous with each other, the "cuda:0"->"cuda:1" copy (`.to()`) can start before the "cpu"->"cuda:0" copy completes (`from_numpy()`).  This requires explicit synchronization, for example you could break it down like this:
```python
wp_a = wp.from_numpy(a)          # cpu->cuda:0 copy
wp.synchronize_device("cuda:0")  # wait for the copy to complete
wp_a = wp_a.to(dev)              # cuda:0->cuda:1 copy
```

But you probably want to skip creating an intermediate array on "cuda:0", so you can pass the `device` argument to `from_numpy()` to place the array on the correct device right away instead of copying it using `.to()`.

```python
wp_a = wp.from_numpy(a, device=dev)
wp_b = wp.from_numpy(b, device=dev)
```

For more info, check out the [concurrency and synchronization](https://nvidia.github.io/warp/modules/concurrency.html) docs.

closed 2024-10-29 by nvlukasz

elrnv, 2024-11-17:
Thank you for the thorough explanation and examples!