NVIDIA/warp #339: [REQUEST] CPU Cloth Simulation Performance

tetosama, 2024-11-04:
### Description
Better performance for CPU Cloth Simulation

### Context
Running example_cloth.py on CPU is hundreds of times slower than that on GPU.

shi-eric, 2024-11-05:
Warp kernels are running single-threaded on the CPU right now, so that is probably the reason for the enormous performance discrepancy. One way we can explore to address this is leveraging OpenMP, but we likely won't have time to explore this until the next calendar year.

korzen, 2025-06-20:
Hi, I have got a simple PBD softbody simulation which runs fine on Warp CUDA in 2ms / step but on Warp CPU it takes 85ms / step. I have the equivalent of this simulation implmented by hand in C/C++ and in Unity using Burst compiler and in both cases it can simulate at below 1ms / step on CPU using single-thread. There is clearly something wrong with CPU kernels performance (tested in Warp 1.7 and 1.8 dev on both Windows Ryzen 5950x, RTX 3700 and on MacMini M1)