NVIDIA/warp #341: [BUG] wp.array Fails to Convert dtype to wp.vec3 without warning

mehdiataei, 2024-11-05:
### Bug Description

Hello!

```
import warp as wp
x = wp.zeros((100, 3), dtype=wp.float32)
print(x.dtype)  # Expected: <class 'warp.types.float32'>

x = wp.array(x, dtype=wp.vec3)
print(x.dtype)  # Expected: <class 'warp.types.vec3'>, but you get: <class 'warp.types.float32'>

```

It may not be a bug inherently, but an error could be raised if the conversion is not possible.

nvlukasz, 2024-11-05:
Thanks Mehdi, this seems like a limitation we should address.

christophercrouzet, 2025-01-18:
Hi @mehdiataei, this has been addressed in commit [a6a8fcd](https://github.com/NVIDIA/warp/commit/a6a8fcd7c91ef6fb3fbf7b95307ac9264df8b566) and will be part of the next release of Warp, sorry for any inconvenience caused!

closed 2025-01-18 by christophercrouzet