NVIDIA/warp #343: [QUESTION] OpenGLRenderer render_mesh render wrong colors

junjeec, 2024-11-06:
Hello,

I'm trying to render a mesh by passing in vertices, triangle indices, and vertex colors using the `OpenGLRenderer'`s `render_mesh` function.
I can see the geometry being rendered correctly, but the colors is off. 

I assume the [colors input to the render_mesh function ](https://github.com/NVIDIA/warp/blob/1d9a0719d11b76162e84c32468081306603052f5/warp/render/render_opengl.py#L2864C9-L2864C15) takes in float array of size (N,3) where N is the number of vertices, and rgb value range between 0-1?!
And the value I get with `renderer.get_pixels(left_color_pixel, split_up_tiles=False, mode="rgb", use_uint8=True)` gives RGB pixel value in range 0-255?

I also tested not sending the colors input to the render_mesh function and get the similar rendered image when passing in my vertices colors. 

Am I missing anything?

Thank you!!

Attaching the color images I rendered with open3d on the same input format, and also the blue mesh I got when rendered with OpenGLRenderer.
[image]
[image]

christophercrouzet, 2024-11-07:
Hi @junjeec, thank you for opening this issue!

The method `OpenGLRenderer.render_mesh()` is currently ignoring any custom colours being passed to it, which is why it is not working as expected. We will have a fix for this in the upcoming release, apologies for any inconvenience caused!

junjeec, 2024-11-07:
Ah I see, thanks for the clarification! 
In the upcoming release, the format I'm using should be correct right? Like it is supposed to take vertex colors (not triangles colors)

Should I close this issue for now? or do you prefer to leave it open?

Thanks!

christophercrouzet, 2024-11-07:
> In the upcoming release, the format I'm using should be correct right? Like it is supposed to take vertex colors (not triangles colors)

Yes, that's right. One colour per point/vertex. Here's a small example I've used to debug the issue:

```py
import numpy as np
import pyglet
import warp as wp
import warp.render

BOX_POINTS = np.array(
    (
        ( 0.853553, -0.146446,  0.0     ), ( 0.146446, -0.853553,  0.0     ),
        ( 0.353553,  0.353553,  0.707106), (-0.353553, -0.353553,  0.707106),
        (-0.353553, -0.353553, -0.707106), ( 0.353553,  0.353553, -0.707106),
        (-0.853553,  0.146446,  0.0     ), (-0.146446,  0.853553,  0.0     ),
    ),
    dtype=np.float32,
)

BOX_FACE_VERTEX_INDICES = np.array(
    (
        0, 3, 1, 0, 2, 3, 4, 7, 5, 4, 6, 7, 6, 2, 7, 6, 3, 2,
        5, 1, 4, 5, 0, 1, 5, 2, 0, 5, 7, 2, 1, 6, 4, 1, 3, 6,
    ),
    dtype=np.int32,
)

resolution = (512, 384)
num_frames = 240
fps = 24
camera_pos = (0.0, 2.5, 5.0)
camera_front = (0.0, -0.5, -1.0)

renderer = warp.render.OpenGLRenderer(
    fps=fps,
    screen_width=resolution[0],
    screen_height=resolution[1],
    camera_pos=camera_pos,
    camera_front=camera_front,
    vsync=True,
)

rng = np.random.default_rng(123)
for frame in range(num_frames):
    color = wp.sin(float(frame) * 0.05) * 0.5 + 0.5

    renderer.begin_frame(frame / num_frames)
    renderer.render_mesh(
        "surface",
        BOX_POINTS,
        BOX_FACE_VERTEX_INDICES,
        colors=((color, 0.5, 0.5),) * len(BOX_POINTS),
    )
    renderer.end_frame()
```

> Should I close this issue for now? or do you prefer to leave it open?

We can leave it open for now—I'll update this issue when the corresponding MR will make it to `main` and we can close it at that time.

Thank you!

junjeec, 2024-11-08:
Got it thanks! 

In addition to support custom vertex color, does warp-lang plan to support texture mesh using uv texture map?

christophercrouzet, 2024-11-17:
Hi @junjeec, we are currently not planning to implement much more features to the OpenGL renderer since making it a fully-fledged renderer is outside the scope of Warp. Its primary intention instead is for debugging purposes. More sophisticated renderings should be done by exporting to a USD file first, and then rendering that scene in an external application.

I've updated the MR to clarify the intended usage of the OpenGL renderer in the documentation.

Thank you for your understanding!

christophercrouzet, 2024-11-21:
Hi @junjeec, just to let you know that this has been implemented in [20c483a](https://github.com/NVIDIA/warp/commit/20c483a178512df4d3e6beb26bd8ddbbbb1361cd) and will be part of the next release of Warp. Thanks again!

junjeec, 2024-11-22:
Thanks very much for the update!!

junjeec, 2024-12-04:
Hello @christophercrouzet ,

I just tested v1.5.0, but the color of the mesh is still not reacting to the color input. Tested with both my data and the example you shared, in your example, the pop up window only shows a static gray cube and I'm not seeing the color being changed. 

Would you point me to what I might be missing? 

Thank you!!

christophercrouzet, 2024-12-08:
Apologies @junjeec, you're not missing anything, what happened is that a follow-up commit added support for rendering meshes with changing topology, and I made sure that the colours were working in that scenario but I didn't check that they were still working with static meshes. I've just opened internally a new MR that addresses this (hopefully) once and for all.

One current limitation is that that per-point colour is not well supported, so I've updated the script above to reflect that.

christophercrouzet, 2024-12-17:
Hi @junjeec, this should _hopefully_ have been fixed in commit [6ab17985](https://github.com/NVIDIA/warp/commit/6ab179855182dfc30c7c3a058be445f583ba09ec), but please do let me know if you're still encountering any issue!