NVIDIA/warp #346: [QUESTION] How to use Autodifferentiation

rrzhang139, 2024-11-06:
Hello, thank you for building this first of all. 

I have a question about differentiating the parameters in Warp. If I wanted to differentiate soft body parameters like k_mu or k_lambda in soft_grid, how would that generally work? Would I initialize some tensors and pass them into kernels to get that working? Do you have any code samples that does this for soft body simulation? Thanks!

amabilee, 2024-11-06:
Hey there !

To differentiate soft body parameters like `k_mu` or `k_lambda` in Warp, you indeed need to initialize tensors and pass them into kernels. Warp supports automatic differentiation, which allows you to compute gradients of these parameters.

Here is the doc about it:
https://nvidia.github.io/warp/modules/differentiability.html

shi-eric, 2024-11-07:
I'm not an expert on the `warp.sim` things, but I'll attempt an answer. You would generally need to do two things:

- When calling the `finalize()` method of the `ModelBuilder` class, you need to use `requires_grad=True` to ensure that gradients are calculated for the `Model` object arrays.
- After recording the forward pass onto a `wp.Tape()` and calling the `backward()` method with the appropriate `loss` input, you would then get the gradients of the `Model.tet_materials` array. The details of this array are here: https://nvidia.github.io/warp/modules/sim.html#warp.sim.Model.tet_materials

rrzhang139, 2024-11-10:
Sorry, I tried implementing your method of computing gradients on `Model.tet_materials` array but it does not seem to be updating. Here is my optimization step:

```
        tape = wp.Tape()
        
        with tape:
            print(f"\nStart of simulation:")
            print(f"target k_mu: {self.target_model.tet_materials.numpy()[0][0]}")
            print(f"k_mu: {self.model.tet_materials.numpy()[0][0]}")
            
            for _ in range(self.num_frames):
                self.step()
            
            print(f"End of simulation:")
            print(f"k_mu: {self.model.tet_materials.numpy()[0][0]}")
            print(f"k_mu gradients: {self.model.tet_materials.grad.numpy()[0][0]}")
            loss = self.compute_loss()
        
        tape.backward(loss)
        tape.zero()
```

The gradients are 0.0, and I compute loss by calculating the MSE between `self.state_0.particle_q` and `self.target_state_0.particle_q`.

Here is my entire script in case you'd like to reproduce it. Thank you so much for your help! 

```
import math
import os
import numpy as np
import torch
import warp as wp
import warp.sim
# from pxr import Usd, UsdGeom

from renderutils import SoftRenderer
from utils.logging import write_imglist_to_dir, write_imglist_to_gif

@wp.kernel
def compute_mse_loss(
    current_pos: wp.array(dtype=wp.vec3),
    target_pos: wp.array(dtype=wp.vec3), 
    loss: wp.array(dtype=float),
):
    # Get thread index
    idx = wp.tid()
    
    # Compute difference for this position
    diff = current_pos[idx] - target_pos[idx]
    squared_dist = wp.dot(diff, diff)
    
    # Atomically add to total loss
    wp.atomic_add(loss, 0, squared_dist)

@wp.kernel
def compute_image_mse_loss(
    rendered_frames: wp.array(dtype=wp.float32),
    target_frames: wp.array(dtype=wp.float32),
    loss: wp.array(dtype=float),
):
    # Get thread index
    idx = wp.tid()
    
    # Compute pixel-wise difference
    diff = rendered_frames[idx] - target_frames[idx]
    squared_diff = diff * diff
    
    # Atomic add to accumulate loss across all pixels
    wp.atomic_add(loss, 0, squared_diff)

[... 305 lines cut ...]
            demo.step()
            rgba = demo.render()
            frames.append(rgba)
        
        # Compute loss and optimize
        loss = demo.optimize_step()

        if args.verbose:
            print(f"Iteration {iteration}, Loss: {loss.numpy()[0]:.6f}")
        
        # Save intermediate results
        if iteration % 10 == 0:
            save_dir = os.path.join(args.logdir, f"iteration_{iteration}")
            os.makedirs(save_dir, exist_ok=True)
            write_imglist_to_gif(frames, os.path.join(save_dir, "sim.gif"), imgformat="rgba")

    # Save final output
    print(f"Saving to {args.logdir}")
    write_imglist_to_dir(frames, os.path.join(args.logdir, "frames"), imgformat="rgba")
    write_imglist_to_gif(frames, os.path.join(args.logdir, "final_sim.gif"), imgformat="rgba")

if __name__ == "__main__":
    import argparse
    main()
```

shi-eric, 2024-11-11:
Hey @rrzhang139, I think your script makes reference to some local code so I can't run it on my system, but one issue I see is that you're only using two model states and swapping them at the end of each `simulate_step()`, which won't work when computing gradients because automatic differentiation requires all intermediate states to be available during the backward pass. See https://github.com/NVIDIA/warp/blob/3f9038d7234b036d7c690314b5a5d7e7ed449e75/warp/examples/optim/example_cloth_throw.py#L104-L107 as an example of creating enough model states so that gradients can be correctly propagated.

Also, in your first snippet, the `.grad` arrays won't be populated because your statements are inside the `Tape()` context. The `Tape.backward()` is what performs the backward pass, so you would inspect gradients _after_ calling `backward()` and _before_ calling `Tape.zero()`.

@eric-heiden for viz

rrzhang139, 2024-11-12:
Thank you @shi-eric, my gradients were flipflopping that indicated the swapped states. The problem I have now is instability and non-convergence. Does warp have an example script on materials prediction? Inherently it seems difficult because you have to track a target model's movement trajectory which introduces a moving optimization target. Also the material parameter may not contain enough information to smoothly update the gradients. 

Let me know what your thoughts are, I have pasted a gist of my updated code. And there should be no dependencies anymore so you can run the script entirely. 

https://gist.github.com/rrzhang139/707d9920be003faf142c32e2ac8e892b

Thanks for your help!

shi-eric, 2024-11-12:
Hey @rrzhang139, I don't have experience with that, but I'll ask around to see if anyone else from the team can weigh in. One of our favorite examples from the community is NCLaw: https://github.com/PingchuanMa/NCLaw Maybe they employ some techniques to get around the issues you are running into.

rrzhang139, 2024-11-12:
Thank you! @shi-eric  One quick question that may help me. One improvement is to scale the gradients in logspace. Is there a way to do forward pass in simulation in regular space then convert to log space for gradient calculation? Would I need to modify the source code?

MontaEllis, 2025-01-07:
Same question. The gradient is 0.

shi-eric, 2025-02-24:
Closing this issue since @MiguelZamoraM  added an example in #419

closed 2025-02-24 by shi-eric