NVIDIA/warp #355: [DOCS] Example error in Differentiability chapter

OneCoin123, 2024-11-14:
## Category

- [x] Report an error in the documentation.
- [ ] Request for something to be documented.
- [ ] Request for a new [example](https://github.com/NVIDIA/warp/tree/main/warp/examples) to illustrate how to do something.
- [ ] Other (please explain)

## Description

Thanks for your excellent work!
During my learning about warp.autograd, I found a wrong example (maybe a typo?) in the document "[Differentiability -> Array Overwrite Tracking -> A: Inter-Kernel Overwrite](https://nvidia.github.io/warp/modules/differentiability.html#array-overwrite-tracking)". The related code is
```Python
tape = wp.Tape()
with tape:
    wp.launch(square_kernel, a.shape, inputs=[a], outputs=[b])
    wp.launch(overwrite_kernel, c.shape, inputs=[c], outputs=[a])
    wp.launch(loss_kernel, a.shape, inputs=[a, loss])

tape.backward(loss)

print(a.grad)
# prints [-2. -4. -6.] instead of [2. 4. 6.]
```
In the given example, array `b` stores the square of array `a`, and `a` is overwritten by `c`, then compute the loss `loss`. When calling `tape.backward` with parameter `loss`, only `loss` has a non-zero initial gradient. Backwarding from `b` to `a` according to the first launch will not accumulate any gradient to `a`, because the adjoint from `b` is zero. As a result, the final printing is [1., 1., 1.] rather than [-2, -4, -6].

As I tried, the `loss` should be computed from `b` to make the output node ('loss') of DAG a successor node of 'b'. When changing the last launch to `wp.launch(loss_kernel, b.shape, inputs=[b, loss])`, the printing works just as the document says.

I think it's just a little typo by mistake, but it might confuse beginners.

shi-eric, 2024-11-14:
Hey @OneCoin123, thank you for reporting this issue! I was adding [doctest](https://www.sphinx-doc.org/en/master/usage/extensions/doctest.html) to some of our code blocks this week and noticed the same issue as you! It was fixed in 28db5de, which was merged just a few minutes after you reported this issue. Please let us know if you find any other mistakes!

OneCoin123, 2024-11-14:
@shi-eric 

Great! What a coincidence haha. Appreciate your nice work! :)

closed 2024-11-15 by shi-eric