NVIDIA/warp #356: [BUG] TestFem.test_integrate_gradient_cpu segfaults on arm64 macOS runner

shi-eric, 2024-11-14:
### Bug Description

https://github.com/NVIDIA/warp/actions/runs/11832532560/job/32970493971 was the first workflow that started failing. By running the tests serially, we see that there's something about `test_integrate_gradient_cpu` that is problematic on the `macos-latest` runner provided by GitHub, which happens to be an arm64 runner.  The problem seems to be avoided if we switch to an x86_64 runner by using the `macos-13` tag...

```
test_grid_2d_cpu (warp.tests.test_fem.TestFem.test_grid_2d_cpu) ...
test_grid_2d_cpu (warp.tests.test_fem.TestFem.test_grid_2d_cpu) ... ok
test_grid_3d_cpu (warp.tests.test_fem.TestFem.test_grid_3d_cpu) ...
test_grid_3d_cpu (warp.tests.test_fem.TestFem.test_grid_3d_cpu) ... ok
test_hex_mesh_cpu (warp.tests.test_fem.TestFem.test_hex_mesh_cpu) ...
test_hex_mesh_cpu (warp.tests.test_fem.TestFem.test_hex_mesh_cpu) ... ok
test_implicit_fields (warp.tests.test_fem.TestFem.test_implicit_fields) ...
test_implicit_fields (warp.tests.test_fem.TestFem.test_implicit_fields) ... ok
test_integrate_gradient_cpu (warp.tests.test_fem.TestFem.test_integrate_gradient_cpu) ...
/Users/runner/work/_temp/ac229663-9565-4bb4-bb78-954bc9168419.sh: line 1:  3798 Segmentation fault: 11  python -m warp.tests --junit-report-xml rspec.xml --coverage --coverage-xml 
```

gdaviet, 2025-06-30:
I finally got hold of an arm64 mac so I could investigate this issue. Unfortunately I have not identified the root cause yet, and not fully eliminated the possibility of a compiler issue.

Here is a smaller repro, independent of warp.fem:
```python 
import warp as wp

wp.clear_kernel_cache()
wp.config.print_launches = True
wp.config.mode = "debug" 

@wp.struct
class Data:
    points : wp.array(dtype=wp.vec3)
    weights : wp.array(dtype=float)

@wp.struct
class MoreData:
    pass

@wp.kernel
def itg(
    qp: Data,
    part:  MoreData,
    part1: MoreData,
    part2: MoreData,
    loss:wp.array(dtype=float)
):
    qp_weight = qp.weights[0]
    wp.atomic_add(loss, 0, qp_weight)

loss = wp.zeros(1, dtype=float, requires_grad=True)

qp_arg = Data()
qp_arg.points = wp.zeros(1, dtype=wp.vec3)
qp_arg.weights = wp.ones(1, dtype=float)

tape = wp.Tape()
with tape:
    wp.launch(itg, dim=1, 
              inputs= [
                  qp_arg,
                  MoreData(),
                  MoreData(),
                  MoreData(),
                  loss
              ])

tape.backward(loss)

print(loss)
print(loss.grad)
```

shi-eric, 2025-06-30:
> I finally got hold of an arm64 mac so I could investigate this issue. Unfortunately I have not identified the root cause yet, and not fully eliminated the possibility of a compiler issue.
> 
> Here is a smaller repro, independent of warp.fem:
> 
> import warp as wp
> 
> wp.clear_kernel_cache()
> wp.config.print_launches = True
> wp.config.mode = "debug" 
> 
> @wp.struct
> class Data:
>     points : wp.array(dtype=wp.vec3)
>     weights : wp.array(dtype=float)
> 
> @wp.struct
> class MoreData:
>     pass
> 
> @wp.kernel
> def itg(
>     qp: Data,
>     part:  MoreData,
>     part1: MoreData,
>     part2: MoreData,
>     loss:wp.array(dtype=float)
> ):
>     qp_weight = qp.weights[0]
>     wp.atomic_add(loss, 0, qp_weight)
> 
> 
> loss = wp.zeros(1, dtype=float, requires_grad=True)
> 
> qp_arg = Data()
> qp_arg.points = wp.zeros(1, dtype=wp.vec3)
> qp_arg.weights = wp.ones(1, dtype=float)
> 
> tape = wp.Tape()
> with tape:
>     wp.launch(itg, dim=1, 
>               inputs= [
>                   qp_arg,
>                   MoreData(),
>                   MoreData(),
>                   MoreData(),
>                   loss
>               ])
> 
> tape.backward(loss)
> 
> print(loss)
> print(loss.grad)

We were suspecting that it might have something having to do with a ctypes calling convention mismatch... And a similar bug appears on aarch64 platforms like Jetson and Grace, also dependent on CPython installation itself as some exhibit the bug and others don't. At this point there are a bunch of issues on the Warp GitHub probably tracking the same thing.

@c0d1f1ed might have ideas

c0d1f1ed, 2025-07-22:
Thanks @gdaviet for the small repro!

My observations thus far: It crashes in `memmove()`, which is called by `libffi`, the native library used by ctypes for calling the binary function compiled from our generated C++ code. This function is called as part of passing the function arguments on the stack, most likely structs.

Interestingly it's not the sheer amount of data. I've tested up to 1000 scalar arguments. Also passing a struct with 1000 members works fine.

I can make Gilles' sample crash in the forward pass by duplicating the arguments. But it works fine when removing any one of the arguments, or when passing `Data`'s members as separate arguments or when replacing `MoreData` with scalars.

Indeed there appears to be a bug in `libffi` for certain combinations of arguments. But it's not clear to me yet what exactly makes it trip up and what it can handle safely. It's worth noting that while the AArch64 calling convention dictates which registers are available for passing arguments, it's up to the platform and language's ABI how those registers are used: https://github.com/ARM-software/abi-aa/releases.

One potential way to work around this is to essentially put everything in a local struct and then pass that struct as a single pointer argument, circumventing the more complex ABI rules.