NVIDIA/warp #361: [QUESTION] <title>

lisiyao21, 2024-11-20:
Hi! Thanks for this great work. I have a humanoid agent (ball joints) and an object. Can current integrator (say XPBD) support the physical between them? I tended to use omni/isaac-sim but they don't support differentiation...

eric-heiden, 2024-11-22:
Hi @lisiyao21,

Yes, XPBD should be able to simulate a humanoid and an object.

lisiyao21, 2024-11-27:
@eric-heiden Thanks for reply! Can I know what is the target of td-controller for joint_6d angles? It seems not to be axis angle. In XPBD integrator code:

```python
        rel_q = wp.quat_inverse(q_p) * q_c

        qtwist = wp.normalize(wp.quat(rel_q[0], 0.0, 0.0, rel_q[3]))
        qswing = rel_q * wp.quat_inverse(qtwist)

        # decompose to a compound rotation each axis
        s = wp.sqrt(rel_q[0] * rel_q[0] + rel_q[3] * rel_q[3])
        invs = 1.0 / s
        invscube = invs * invs * invs

        # handle axis-angle joints

        # rescale twist from quaternion space to angular
        err_0 = 2.0 * wp.asin(wp.clamp(qtwist[0], -1.0, 1.0))
        err_1 = qswing[1]
        err_2 = qswing[2]
``` 

When setting joint_act, should I first transform the rotvec (axis angle) to such decomposition? thanks!

closed 2025-11-06 by shi-eric